nasa/fprime #2398: Generate Failed. New FPPv2.0 syntax error?

ethancheez, 2023-11-29:
| | |
|:---|:---|
|**_F´ Version_**| v3.4.0 |
|**_Affected Component_**|   |
---
## Problem Description

See Error after running `fprime-util generate`:
```
fprime-util generate
[INFO] Generating build directory at: /home/ethan/work/fprime-projects/fprime-community/fprime-tutorial-hello-world/build-fprime-automatic-native
[INFO] Using toolchain file None for platform default
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
/home/ethan/venv/fprime-new/bin/fpp-depend: 1: Syntax error: "(" unexpected
-- [fpp-tools] fpp-depend appears corrupt.
CMake Error at fprime/cmake/required.cmake:26 (message):
-- Configuring incomplete, errors occurred!
   fpp tools not found. Install with:
    'pip install -r "/home/ethan/work/fprime-projects/fprime-community/fprime-tutorial-hello-world/fprime/requirements.txt"'
Call Stack (most recent call first):
  fprime/cmake/FPrime.cmake:13 (include)
  CMakeLists.txt:13 (include)
```

## How to Reproduce

The following commands were executed in order:
```
pip install fprime-tools
pip install -r "fprime/requirements.txt"
fprime-util generate
```

thomas-bc, 2023-11-29:
@ethancheez I haven't been able to reproduce your issue. Do you have more info/context? What about running 
```
pip install -Ur fprime/requirements
```
to make sure packages have been upgraded?

Or maybe try with a fresh virtual environment.

ethancheez, 2023-11-29:
Yes I have made an entirely new virtual environment, and also added the `-Ur` flag. Same error

thomas-bc, 2023-11-29:
And you are using hello-world tutorial v3.4.0 correct?
Can you let me know what host/OS you're using, and version of Python?

Could you also paste the result of `pip freeze | grep fprime`

ethancheez, 2023-11-29:
Running Windows 11, WSL Ubuntu 22.04.
Python 3.10.12

```
fprime-fpl-convert-xml==1.0.1a4
fprime-fpl-extract-xml==1.0.1a4
fprime-fpl-layout==1.0.1a4
fprime-fpl-write-pic==1.0.1a4
fprime-fpp-check==2.0.1
fprime-fpp-depend==2.0.1
fprime-fpp-filenames==2.0.1
fprime-fpp-format==2.0.1
fprime-fpp-from-xml==2.0.1
fprime-fpp-locate-defs==2.0.1
fprime-fpp-locate-uses==2.0.1
fprime-fpp-syntax==2.0.1
fprime-fpp-to-cpp==2.0.1
fprime-fpp-to-json==2.0.1
fprime-fpp-to-xml==2.0.1
fprime-gds==3.4.0
fprime-tools==3.4.0
fprime-visual==1.0.1
```

jsilveira1409, 2023-12-03:
I'm having the exact same issue on a fresh install of fprime, however, it is on Ubuntu 22 :(

thomas-bc, 2023-12-04:
Dumping some investigate notes in here, unsure whether it is related just yet:

One of the latest successful run building fpp native images https://github.com/fprime-community/fpp/actions/runs/6459740810/job/17540152759 includes this warning. If not related to the current issue, it most certainly is related to the recent fails we've seen in the build
```
        ********************************************************************************
        The arguments ['--plat-name=manylinux_2_28_x86_64'] were given via `--global-option`.
        Please use `--build-option` instead,
        `--global-option` is reserved for flags like `--verbose` or `--quiet`.

        This deprecation is overdue, please update your project and remove deprecated
        calls to avoid build errors in the future.
        ********************************************************************************
```

thomas-bc, 2023-12-04:
@ethancheez @jsilveira1409 would you be able to let me know which version of Python and pip you are running?
```
python --version && pip --version
```

I managed to get a similar issue on WSL2, although it's not exactly the same one so I'm not sure if this will help. Instead of the fpp-depend script being corrupted, it wasn't being installed at all. This was due to me having a too old version of pip on my machine.
Running `pip install --upgrade pip` did the trick, and I'm able to run `fprime-util generate` on WSL2 with F´ v3.4.0.

Can you confirm when the `fprime-fpp-depend` package gets installed, which wheel it's using?
It should read something like
```
fprime_fpp_depend-2.0.1-py3-none-manylinux_2_28_x86_64.whl
```
and **not**
```
fprime_fpp_depend-2.0.1-py3-none-any.whl
```

ethancheez, 2023-12-04:
```
Python 3.10.12
pip 22.0.2 from /usr/lib/python3/dist-packages/pip (python 3.10)
```

```
Collecting fprime-fpp-depend==2.0.1
  Using cached fprime_fpp_depend-2.0.1-py3-none-manylinux_2_28_x86_64.whl (9.2 MB)
```

thomas-bc, 2023-12-04:
What happens if you run `fpp-depend --help` ?

ethancheez, 2023-12-04:
```
fpp-depend v2.0.1
Usage: fpp-depend [options] [file ...]

  -h, --help               print this message and exit
  -a, --auto-test-helpers  enable automatic generation of test helper code
  -d, --direct <file>      write direct dependencies to file
  -f, --framework <file>   write framework dependencies to file
  -g, --generated <file>   write names of generated autocode files to file
  -i, --included <file>    write included dependencies to file
  -m, --missing <file>     write missing dependencies to file
  -u, --unit-test <file>   write names of generated unit test files to file
  file ...                 input files
```

thomas-bc, 2023-12-04:
Ohhh wot

Then maybe all you need is `fprime-util generate -DFPRIME_SKIP_TOOLS_VERSION_CHECK=ON`, this will skip the [tool check](https://github.com/nasa/fprime/blob/0ed8e0b74e2cc99e519576f6cfc250e7d21ca10c/cmake/autocoder/fpp.cmake#L36-L48). I'm not sure why it's tripping...

thomas-bc, 2023-12-04:
Could you try opening `TestProject/fprime/cmake/target/tools/redirector.py` and adding at the very first line:

```
#!/usr/bin/env python3
```

ethancheez, 2023-12-04:
> Could you try opening `TestProject/fprime/cmake/target/tools/redirector.py` and adding at the very first line:
> 
> ```
> #!/usr/bin/env python3
> ```

This results in the same error.

thomas-bc, 2023-12-04:
Yeah nevermind we're actually running the file using python ..., not as an executable.

Was reading through [this](https://stackoverflow.com/a/47329417), can you confirm the ouput of

```
ls -lh <path/to/venv>/bin/fp*
```

ethancheez, 2023-12-04:
```
-rwxr-xr-x 1 root root 20M Dec  2 14:14 /root/venv/fprime/bin/fpl-convert-xml
-rwxr-xr-x 1 root root 239 Dec  2 14:14 /root/venv/fprime/bin/fpl-extract-xml
-rwxr-xr-x 1 root root 15M Dec  2 14:14 /root/venv/fprime/bin/fpl-layout
-rwxr-xr-x 1 root root 15M Dec  2 14:14 /root/venv/fprime/bin/fpl-write-pic
-rwxr-xr-x 1 root root 34M Dec  2 14:14 /root/venv/fprime/bin/fpp-check
-rwxr-xr-x 1 root root 33M Dec  2 14:14 /root/venv/fprime/bin/fpp-depend
-rwxr-xr-x 1 root root 33M Dec  2 14:14 /root/venv/fprime/bin/fpp-filenames
-rwxr-xr-x 1 root root 32M Dec  2 14:14 /root/venv/fprime/bin/fpp-format
-rwxr-xr-x 1 root root 37M Dec  2 14:14 /root/venv/fprime/bin/fpp-from-xml
-rwxr-xr-x 1 root root 32M Dec  2 14:14 /root/venv/fprime/bin/fpp-locate-defs
-rwxr-xr-x 1 root root 34M Dec  2 14:14 /root/venv/fprime/bin/fpp-locate-uses
-rwxr-xr-x 1 root root 32M Dec  2 14:14 /root/venv/fprime/bin/fpp-syntax
-rwxr-xr-x 1 root root 38M Dec  2 14:14 /root/venv/fprime/bin/fpp-to-cpp
-rwxr-xr-x 1 root root 34M Dec  2 14:14 /root/venv/fprime/bin/fpp-to-json
-rwxr-xr-x 1 root root 36M Dec  2 14:14 /root/venv/fprime/bin/fpp-to-xml
-rwxr-xr-x 1 root root 248 Dec  2 14:14 /root/venv/fprime/bin/fprime-cli
-rwxr-xr-x 1 root root 252 Dec  2 14:14 /root/venv/fprime/bin/fprime-gds
-rwxr-xr-x 1 root root 245 Dec  2 14:14 /root/venv/fprime/bin/fprime-seqgen
-rwxr-xr-x 1 root root 235 Dec  2 14:08 /root/venv/fprime/bin/fprime-util
-rwxr-xr-x 1 root root 237 Dec  2 14:08 /root/venv/fprime/bin/fprime-version-check
-rwxr-xr-x 1 root root 237 Dec  2 14:14 /root/venv/fprime/bin/fprime-visual
```

thomas-bc, 2023-12-04:
And what about `fpp-locate-defs --help` ?

FWIW, I was able to get that same error message on my Mac by overwriting the script `bin/fpp-locate-defs` with an empty file. 
Have you tried running in a virtual environment outside of root?

ethancheez, 2023-12-04:
Running a new venv outside of root and `fpp-locate-defs --help` results in a formatting error. 
```sh
-bash: /venv/fprime/bin/fpp-locate-defs: cannot execute binary file: Exec format error
```

thomas-bc, 2023-12-04:
Does the entire build process succeed on WSL2?

thomas-bc, 2023-12-04:
And this may be a dummy question, but did prior versions of F´ ever work with WSL1?

ethancheez, 2023-12-04:
> And this may be a dummy question, but did prior versions of F´ ever work with WSL1?

Yes I have been using WSL1 (when I started working with you guys in May), and have been using it up until the release of v3.4.0.

ethancheez, 2023-12-04:
For reference, here are the results of running the same commands on WSL1 using previous versions of fprime and fpp:

```sh
# python --version && pip --version
Python 3.10.12
pip 22.0.2 from /venv/fprime3.3.2/lib/python3.10/site-packages/pip (python 3.10)
```

Using [`fprime@ba8f1c6ac`](https://github.com/nasa/fprime/commit/ba8f1c6ac8ca66e860d1c48b3eccee12ee10d7e0) (release v3.3.2)

```sh
# pip freeze | grep fprime
fprime-fpl-convert-xml==1.0.0
fprime-fpl-extract-xml==1.0.0
fprime-fpl-layout==1.0.0
fprime-fpl-write-pic==1.0.0
fprime-fpp==1.3.0
fprime-gds==3.3.2
fprime-tools==3.3.3
fprime-visual==1.0.0
```

```sh
# fpp-depend --help
fpp-depend v1.3.0
Usage: fpp-depend [options] [file ...]

  -d, --direct <file>     write direct dependencies to file
  -f, --framework <file>  write framework dependencies to file
  -g, --generated <file>  write names of generated autocode files to file
  -i, --included <file>   write included dependencies to file
  -m, --missing <file>    write missing dependencies to file
  -h, --help              print this message and exit
  -t, --template <file>   write names of generated template files to file
  file ...                input files
```

```sh
# fpp-locate-defs --help
fpp-locate-defs v1.3.0
Usage: fpp-locate-defs [options] [file ...]

  -d, --directory <dir>  base directory
  -h, --help             print this message and exit
  file ...               files to analyze
```

```sh
# ls -lh /venv/fprime3.3.2/bin/fp*
-rwxr-xr-x 1 root root 20M Dec  4 12:01 /venv/fprime3.3.2/bin/fpl-convert-xml
-rwxr-xr-x 1 root root 235 Dec  4 12:01 /venv/fprime3.3.2/bin/fpl-extract-xml
-rwxr-xr-x 1 root root 15M Dec  4 12:01 /venv/fprime3.3.2/bin/fpl-layout
-rwxr-xr-x 1 root root 15M Dec  4 12:01 /venv/fprime3.3.2/bin/fpl-write-pic
-rwxr-xr-x 1 root root 21M Dec  4 12:01 /venv/fprime3.3.2/bin/fpp-check
-rwxr-xr-x 1 root root 18M Dec  4 12:01 /venv/fprime3.3.2/bin/fpp-depend
-rwxr-xr-x 1 root root 18M Dec  4 12:01 /venv/fprime3.3.2/bin/fpp-filenames
-rwxr-xr-x 1 root root 18M Dec  4 12:01 /venv/fprime3.3.2/bin/fpp-format
-rwxr-xr-x 1 root root 23M Dec  4 12:01 /venv/fprime3.3.2/bin/fpp-from-xml
-rwxr-xr-x 1 root root 18M Dec  4 12:01 /venv/fprime3.3.2/bin/fpp-locate-defs
-rwxr-xr-x 1 root root 21M Dec  4 12:01 /venv/fprime3.3.2/bin/fpp-locate-uses
-rwxr-xr-x 1 root root 18M Dec  4 12:01 /venv/fprime3.3.2/bin/fpp-syntax
-rwxr-xr-x 1 root root 22M Dec  4 12:01 /venv/fprime3.3.2/bin/fpp-to-cpp
-rwxr-xr-x 1 root root 21M Dec  4 12:01 /venv/fprime3.3.2/bin/fpp-to-xml
-rwxr-xr-x 1 root root 244 Dec  4 12:01 /venv/fprime3.3.2/bin/fprime-cli
-rwxr-xr-x 1 root root 248 Dec  4 12:01 /venv/fprime3.3.2/bin/fprime-gds
-rwxr-xr-x 1 root root 241 Dec  4 12:01 /venv/fprime3.3.2/bin/fprime-seqgen
-rwxr-xr-x 1 root root 231 Dec  4 12:01 /venv/fprime3.3.2/bin/fprime-util
-rwxr-xr-x 1 root root 233 Dec  4 12:01 /venv/fprime3.3.2/bin/fprime-version-check
-rwxr-xr-x 1 root root 233 Dec  4 12:01 /venv/fprime3.3.2/bin/fprime-visual
```

jsilveira1409, 2023-12-09:
Got caught up on some work. On Ubuntu 20.04 i'm getting:

```[INFO] Generating build directory at: /home/dphi/Desktop/FS/ClusterGateV2/build-fprime-automatic-aarch64-linux
[INFO] Using toolchain file /home/dphi/Desktop/FS/ClusterGateV2/fprime/cmake/toolchain/aarch64-linux.cmake for platform aarch64-linux
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Check for working C compiler: /opt/toolchains/bin/aarch64-none-linux-gnu-gcc
-- Check for working C compiler: /opt/toolchains/bin/aarch64-none-linux-gnu-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /opt/toolchains/bin/aarch64-none-linux-gnu-g++
-- Check for working CXX compiler: /opt/toolchains/bin/aarch64-none-linux-gnu-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [fpp-tools] Could not find fpp-depend
CMake Error at fprime/cmake/required.cmake:26 (message):
-- Configuring incomplete, errors occurred!
   fpp tools not found. Install with:
See also "/home/dphi/Desktop/FS/ClusterGateV2/build-fprime-automatic-aarch64-linux/CMakeFiles/CMakeOutput.log".
    'pip install -r "/home/dphi/Desktop/FS/ClusterGateV2/fprime/requirements.txt"'
Call Stack (most recent call first):
  fprime/cmake/FPrime.cmake:13 (include)
  CMakeLists.txt:13 (include)
```
  
  running on a venv with the tools already installed. `fpp-depend` does not seem to be found when running `fpp-depend --help`  even after running `pip install -r requirements.txt` and successfully installing it.
  
  currently running: 
  
```  
Python 3.8.10
pip 23.3.1 from prime-venv/lib/python3.8/site-packages/pip (python 3.8)
```
  
`pip freeze | grep fprime` outputs:

```
fprime-fpl-convert-xml==1.0.1a4
fprime-fpl-extract-xml==1.0.1a4
fprime-fpl-layout==1.0.1a4
fprime-fpl-write-pic==1.0.1a4
fprime-fpp-check==2.0.1
fprime-fpp-depend==2.0.1
fprime-fpp-filenames==2.0.1
fprime-fpp-format==2.0.1
fprime-fpp-from-xml==2.0.1
fprime-fpp-locate-defs==2.0.1
fprime-fpp-locate-uses==2.0.1
fprime-fpp-syntax==2.0.1
fprime-fpp-to-cpp==2.0.1
fprime-fpp-to-json==2.0.1
fprime-fpp-to-xml==2.0.1
fprime-gds==3.4.0
fprime-tools==3.4.0
fprime-visual==1.0.1
``` 
  
  
  and yes fpp-depend is installed from the wheel you mentionned @thomas-bc (`fprime_fpp_depend-2.0.1-py3-none-manylinux_2_28_x86_64.whl`)

LeStarch, 2023-12-12:
This looks like v3.4.0 has some bad packages.  The new packages I made today for (prep for v2.0.2 fpp) work just fine.

LeStarch, 2023-12-12:
The only substantial difference I can see is that the new packages are compatible with an older kernel (2.6.32 not 3.2.0).  They also have subtly different sizes....but not of this would cause issue in my estimation.

ethancheez, 2023-12-12:
> This looks like v3.4.0 has some bad packages. The new packages I made today for (prep for v2.0.2 fpp) work just fine.

Any way for me to test this on my WSL1 instance to see if it works? Can I clone [`fprime-tools`](https://github.com/fprime-community/fprime-tools) and just run `pip install -U .` ?

LeStarch, 2023-12-12:
At this point, I will recommend we release the hot fix, v2.0.2 FPP, and test those packages on WSL 1.  If it works, then I'd argue this is a fluke unless we find more corrupt packages.

Unless someone knows how to pin-point what the format error actually is.  The `file` output on the working and not working builds indicate that the file type should be compatible...so it is corruption somewhere in the file, or specific incompatibilities with some platforms.  Both are notoriously hard to reproduce.

LeStarch, 2023-12-12:
This error also implies we should track a checksum of each binary from the time of creation.  Just so we can check to see if something changed during packaging.

LeStarch, 2023-12-12:
> > This looks like v3.4.0 has some bad packages. The new packages I made today for (prep for v2.0.2 fpp) work just fine.
> 
> Any way for me to test this on my WSL1 instance to see if it works? Can I clone [`fprime-tools`](https://github.com/fprime-community/fprime-tools) and just run `pip install -U .` ?

Not easily.  These are interim packages that are built as part of the normal build...but only maintainers have access to the artifacts. However, we are about to do the point release and you could test that!

LeStarch, 2023-12-13:
@ethancheez you should now be able to install via `pip install fprime-fpp==v2.0.2a1` and test the new corrected packages.  Note: this uses our new meta-package for ease of installation of the whole suite.  Make sure all the individual tool packages update.

Also, you'll need to bump fpp versions in requirements.txt or run with -DFPRIM_SKIP_TOOLS_VERSION_CHECK=ON

ethancheez, 2023-12-13:
> @ethancheez you should now be able to install via `pip install fprime-fpp==v2.0.2a1` and test the new corrected packages. Note: this uses our new meta-package for ease of installation of the whole [...]

Successfully generates and builds for native Linux on WSL1!

LeStarch, 2023-12-14:
Wooo!  I am going to close this issue as it seems we fixed this.

closed 2023-12-14 by LeStarch