nasa/fprime #2404: add support for building documentation locally

capsulecorplab, 2023-12-01:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

As per discussion in https://github.com/nasa/fprime/discussions/2364#discussioncomment-7527518 and attempts to build documentation locally as part of the fprime-workspace-image in https://github.com/capsulecorplab/fprime-workspace-image/pull/35, it appears the current fprime/docs codebase isn't suited for a local build of the docs, s.a., the tutorials. This feature/issue is for capturing the need to refactor the current docs codebase to support a local build of the docs.

## Rationale

I'm currently part of an upcoming Mars analog astronaut mission where we're exploring the possible usage of fprime for collecting various telemetry data (s.a., for monitoring vitals in EVA's or plants in a GreenHab). A local build of the documentation is necessary, as it would be considered "breaking sim" if we were to depend on an internet connection to read the fprime documentation.

LeStarch, 2024-04-01:
@ashleynilo assigned to you because you are working on the whole webpage rework task!  I don't expect you to do this work, so much as track it so we can revisit once we have a new setup.

thomas-bc, 2024-04-01:
Linking in https://github.com/nasa/fprime/discussions/2635 as it could be very relevant to this issue

LeStarch, 2025-01-15:
This works with `mkdocs`.

closed 2025-01-15 by LeStarch