nasa/fprime #2411: Add Hotfix CI Branches

LeStarch, 2023-12-07:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Add specific branches (`release/*`) in addition to devel in CI.  This allows us to create hotfixs that don't merge with devel before they are released.

LeStarch, 2023-12-07:
To exclude `release/documentation` we probably should follow the pattern: `release/v*.*.*-hotfix" instead

thomas-bc, 2023-12-07:
It should be doable to do something like this
```
on:
  push:
    branches: [ devel , release/*]
    branches-ignore: [ release/documentation ]
```

https://docs.github.com/en/actions/using-workflows/workflow-syntax-for-github-actions#example-excluding-branches-and-tags