nasa/fprime #2429: Record packet headers

rrieber, 2023-12-11:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

AMPCS has a system that records packet headers. One can do `chill_get_packets` and get a list of all packets received. FPrime-GDS records the raw received data in `recv.bin`, which is nice, but to suck out the packet headers, one has to read this in binary, which is just cruel.

## Rationale

Make my life easier when it comes to monitoring/watching/debugging the downlink.

LeStarch, 2024-01-10:
We should document the process to handle these files so that user may make any flavor of process script they'd like.

LeStarch, 2024-01-10:
Here is the sample GDS script that we will expand for this documentation: https://github.com/fprime-community/fprime-gds-client-sample-script

LeStarch, 2024-03-27:
I am closing this, #2478 will provide the flexibility for users to define scripts for this process.