nasa/fprime #2463: Purge Fails Silently

LMPS97, 2024-01-02:
| | |
|:---|:---|
|**_F´ Version_**| 3.0.0 |
|**_Affected Component_**| fprime-util purge  |
---
## Problem Description

`fprime-util purge` fails silently when it doesn't have permissions to modify a file.

## How to Reproduce

1. Create a F' build using `fprime-util build`
2. Change the permission of a file to not allow editing
3. Run `fprime-util purge`

Or 

1. Create a F' build using `sudo fprime-util build`
2. Run `fprime-util purge`

## Expected Behavior

`fprime-util purge` should instead output an error notifying the user that it doesn't have permission to modify the file.

LeStarch, 2024-01-04:
Interesting catch!  Thanks @LMPS97!

thomas-bc, 2024-01-10:
If new contributors want to try and fix this, here is probably the place to look at first: https://github.com/fprime-community/fprime-tools/blob/devel/src/fprime/fbuild/cli.py

thomas-bc, 2024-02-13:
Thank you @IvoBrandao !

closed 2024-02-13 by thomas-bc