nasa/fprime #2474: F Prime GDS - problems scrolling through channel selection for custom view

timcanham, 2024-01-09:
| | |
|:---|:---|
|**_F´ Version_**|v3.4.1, commit https://github.com/nasa/fprime/commit/8a1612c0dd1ec18ada8162d92d37ff7317835129 |
|**_Affected Component_**|  n/a |
---
## Problem Description

When editing the telemetry view, scrolling back through the list of channels for selection stops working at a certain point. 

## How to Reproduce

1. Go to telemetry tab (needs large number of channels defined)
2. Scroll through list of channels to select
3. At a certain point, the user cannot scroll any further, and the view will jump back to the previous block of channels.

## Expected Behavior

User should be able to scroll and select all the channels.

ljreder, 2024-01-25:
@timcanham I attempted to reproduce this in the simple HelloWorld example but I assume it did not have enough channels.  How many channels did this happen with.  Do you have a case I could try.  Perhaps the old Ref application is floating around some place and that would work.  Please advise.

timcanham, 2024-01-25:
Yes, it happens on an internal project with lots of channels on the order of several hundred. Contact me and I can set up a demo.

LeStarch, 2024-11-25:
I was not able to duplicate this.  I set up nearly 8000 channels and saw no issues.  If I can see a demo of the problem, perhaps we can fix it.  Otherwise, I will close as could not reproduce.

LeStarch, 2024-11-25:
Closing, cannot reproduce.  Reopen or comment if it can be reproduced.

closed 2024-11-25 by LeStarch