nasa/fprime #2477: Create standalone bootstrap tool

thomas-bc, 2024-01-10:
| | |
|:---|:---|
|**_F´ Version_**| v3.4.0 |
|**_Affected Component_**|  |
---
## Feature Description

Task list:

- [ ] Create repo
- [ ] Publish to PyPi
- [ ] Rewrite Install Guide
- [ ] Double check docs for other references of installs

Standalone tool shall:
- same functionality as `fprime-util new --project`
- create a venv within the newly created project, so that package management is similar to how npm does it (everything is within the project)

## Rationale

Make the whole install process more clear, and better venv management

thomas-bc, 2024-01-10:
Also see https://github.com/nasa/fprime/issues/2251

capsulecorplab, 2024-01-16:
I take it the venv creation will still be optional?

thomas-bc, 2024-01-16:
Yes! But the venv will be created within the created project instead of having the user supply one. All dependencies will be contained in the project, similar to how `npm` does it. We're hoping this will make venv management much easier for users who might not be familiar with Python venv. Feel free to suggest any other improvements/approaches 🚀

capsulecorplab, 2024-01-16:
I think you already know my preferred approach, which is building projects in the [fprime-workspace-image](https://github.com/fprime-community/fprime-workspace-image) :upside_down_face: Otherwise, seems like a reasonable approach for creating a venv on the fly

closed 2024-02-16 by thomas-bc