nasa/fprime #2478: Add plug-in approach for packet descriptors

timcanham, 2024-01-10:
| | |
|:---|:---|
|**_F´ Version_**|>= 3.x |
|**_Affected Component_**| n/a |
---
## Feature Description

Provide a simple method (like a directory) for users to supply handlers for non-standard F Prime packet descriptors.

## Rationale

A user might wish to generate their own packet descriptor for special handling.

LeStarch, 2024-03-12:
We now have a plugin system in the GDS that allows users to build a plugin.  One of the plugins we will soon define is "start application" which allows users to plug-in new applications to run!

It won't be as simple as packet_handler_... but it will be close and much more flexible.

LeStarch, 2024-03-12:
We could, theoretically, make the data consumers a plug-in too.....but that would be fairly limited and could make the GDS less-stable.  I think it cleaner to plug in new client apps that consume and use data as they see fit.

LeStarch, 2024-03-27:
We want packet-descriptor handlers easier to write.  This could be done in one of two ways:

1. A subclass of Startup that streamlines the addition of specifically packet handlers (e.g. instantiates StandardPipeline) for you.
2. A direct handler plugin.

closed 2025-06-11 by LeStarch