nasa/fprime #2484: `fprime-util new --project` doesn't prompt for fprime version and creates an fprime submodule checked out at `devel` as of `fprime-tools` v3.40

capsulecorplab, 2024-01-15:
| | |
|:---|:---|
|**_F´ Version_**| 3.4.0 |
|**_Affected Component_**|  |
---
## Problem Description

Haven't tested to see if the same behavior exists in fprime-tools v3.4.1, but at least in v3.4.0, `fprime-util new --project` creates an `fprime` submodule checked out at a grafted `HEAD` commit at `devel` without prompting for fprime version as described in [Creating a New F' Project](https://github.com/fprime-community/fprime-tutorial-hello-world/blob/main/docs/NewProject.md#creating-a-new-f-project); I'd have to manually `cd` into the `fprime` submodule, run `git fetch --tags` followed by `git checkout v3.4.0`

[image]

[image]

## How to Reproduce

1. `pip install fprime-tools==3.4.0`
2. `fprime-util new --project` # enter `MyFPrimeProject` for project name
3. `cd MyFPrimeProject/fprime`
4. `git show HEAD`

## Expected Behavior

`fprime-util new --project` should prompt for fprime version

thomas-bc, 2024-01-16:
Thanks for reporting this bug - this is indeed something that slipped through our testing. Opened https://github.com/fprime-community/fprime-tools/pull/191 to fix it

Not prompting for a version is intended though, we made the decision for the tool to checkout the latest release only. This makes its usage simpler, and still allows experimented users to checkout earlier versions if needed.

thomas-bc, 2024-01-16:
Also just FYI, we're planning in the near future to take `fprime-util new --project` out of `fprime-util` and into a standalone bootstrap tool -- see https://github.com/nasa/fprime/issues/2477

capsulecorplab, 2024-01-16:
> Thanks for reporting this bug - this is indeed something that slipped through our testing. Opened [fprime-community/fprime-tools#191](https://github.com/fprime-community/fprime-tools/pull/191) to fix it
> 
> Not prompting for a version is intended though, we made the decision for the tool to checkout the latest release only. This makes its usage simpler, and still allows experimented users to checkout earlier versions if needed.

I'm confused, so is it by design that `fprime-tools` isn't prompting for a branch or tag now or is it a bug? I kind of preferred the v3.3.2 behavior when it did prompt for a branch or tag (albeit, with the default option was set to `devel`)

thomas-bc, 2024-01-16:
It is by design. It was causing issues here with `fprime-tools` rewriting itself during execution (fprime-util new --project installs in current venv), among other things. We may reconsider when we have the new bootstrap tool as this will alleviate most of the concerns.

closed 2024-01-18 by LeStarch