nasa/fprime #2486: fprime-util generate displays generic error when library is missing target restrictions

kbotteon, 2024-01-16:
| | |
|:---|:---|
|**_F´ Version_**| 3.4.0 |
|**_Affected Component_**| fprime-util |
---
## Problem Description

@LeStarch, per our conversation earlier, you requested a bug report with the following notes:
- fprime-util generate <target> will produce a generic error when one of the subrepos listed as a `library_location` in `settings.ini` is missing the correct `restrict_platforms()` verbiage in its any of its CMake files.
- One example is that in which a library is added to access one or several files, but a target-specific static library included as a `MOD_DEP` is also specified 

## How to Reproduce

1. Include as a `library_location` any submodule or folder tree which includes a CMake that uses a path in a `cmake/toolchain` file
2. Alternatively, set a `MOD_DEPS` that is satisfied by a lower-level CMake file with a `restrict_platforms()` that does not match the higher-level CMake file
3. Observe an error like:

`CMake error at fprime/cmake/target/build.cmake:75 (add_dependencies): The dependency myLib.a of target MyTarget does not exist`

## Expected Behavior

Reading the CMake call/error stack will eventually get you to a line pointing to the underlying offending CMake file. I'm not sure how you would like to improve this.

LeStarch, 2024-01-17:
What is happening is that `restrict_platforms` can remove a library from the build which is a dependency of another library.  This causes the above generic error.

This can be fixed by adding code to track available by restricted targets, and alerting the user when selected as a dependency.

LeStarch, 2024-03-28:
Note: this can be fixed partially.  It can be fixed, but only if the restricted platform comes first in the build order.  Otherwise this cannot be guaranteed.  Since this is most cases, I'll make the fix.

commit 558ad475 (2024-03-28): Produce error on restricted targets.  Fixes: #2486 (#2633)

* Produce error on restricted targets.  Fixes: #2486

* sp
closed 2024-03-28 by LeStarch