nasa/fprime #2488: If cpp files depend on fpp but area listed in reverse order, compile fails

kbotteon, 2024-01-17:
| | |
|:---|:---|
|**_F´ Version_**| 3.4.0 |
|**_Affected Component_**| fprime-util |
---
## Problem Description

This order is required if the cpp file(s) depend on class(es) that fpp generates.

set(SOURCE_FILES
    "${CMAKE_CURRENT_LIST_DIR}/FrameHeaderSerializable.fpp"
    "${CMAKE_CURRENT_LIST_DIR}/MyComponent.fpp"
    "${CMAKE_CURRENT_LIST_DIR}/MyComponent.cpp"
    "${CMAKE_CURRENT_LIST_DIR}/MyComponent.hpp"
)

## How to Reproduce

## Expected Behavior

The build system should auto-code what is required first.

LeStarch, 2024-01-17:
To work around, sort your SOURCE_FILES list with any autocoder model file listed first.

kbotteon, 2024-03-28:
It's been a minute, so I don't quite remember, but you and I were debugging this in 301. I *think* if you #include a header that is auto-coded in your CPP, like a type that is defined in an FPP, and that FPP is listed after the CPP in CMakeLists, the first build will fail on the CPP as the tools haven't yet gotten around to generating the required header.

LeStarch, 2024-03-28:
I have tried the following to reproduce with the `.cpp` file first in the source list:

1. Import component `Ac.hpp` directly in component `.cpp`: No error
2. Import type `Ac.hpp` in component `.cpp`: No error
3. Item 2, but with a separate `.fpp` file: No error
4. Adding type inside component and importing as above: No error.

I hate to say it, but I am unable to reproduce this error.  We can open this back up if it happens again and try pin it down.

closed 2024-03-28 by LeStarch