nasa/fprime #2504: fprime-util generate fails due to info-cache/fpp_locs error

dury2379, 2024-01-27:
---
## Problem Description
I am trying to generate a new empty project and get this error when running fprime-util generate command:

$ fprime-util generate
[INFO] Generating build directory at: /home/den/Documents/F'/HelloWorld/Hello_World/build-fprime-automatic-native-ut
[INFO] Using toolchain file None for platform default
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Enabled the following sanitizers: address,undefined,leak
-- [python3] python3 found at: /home/den/Documents/F'/HelloWorld/fprime-venv/bin/python3
-- [fpp-tools] fpp-depend found at: /home/den/Documents/F'/HelloWorld/fprime-venv/bin/fpp-depend
-- [fpp-tools] fpp-to-xml found at: /home/den/Documents/F'/HelloWorld/fprime-venv/bin/fpp-to-xml
-- [fpp-tools] fpp-to-cpp found at: /home/den/Documents/F'/HelloWorld/fprime-venv/bin/fpp-to-cpp
-- [fpp-tools] fpp-locate-defs found at: /home/den/Documents/F'/HelloWorld/fprime-venv/bin/fpp-locate-defs
-- [FPRIME] Module locations: /home/den/Documents/F'/HelloWorld/Hello_World/fprime;/home/den/Documents/F'/HelloWorld/Hello_World
-- [FPRIME] Configuration module: /home/den/Documents/F'/HelloWorld/Hello_World/fprime/config
-- [FPRIME] Installation directory: /home/den/Documents/F'/HelloWorld/Hello_World/build-artifacts
-- Target build toolchain/platform: Linux/Linux
-- Including /home/den/Documents/F'/HelloWorld/Hello_World/fprime/cmake/platform/Linux.cmake
-- Requiring thread library
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- [sub-build] Performing sub-build: info-cache
-- [sub-build] Generating: info-cache with target/fpp_locs;target/fpp_depend
-- [sub-build] Executing: info-cache with fpp_locs
CMake Error at fprime/cmake/utilities.cmake:586 (message):
-- Configuring incomplete, errors occurred!
  [sub-build] Failed to execute: info-cache/fpp_locs:
See also "/home/den/Documents/F'/HelloWorld/Hello_World/build-fprime-automatic-native-ut/CMakeFiles/CMakeOutput.log".

  /bin/sh: 1: Syntax error: Unterminated quoted string

  gmake[3]: *** [CMakeFiles/fpp_locs.dir/build.make:74: ../locs.fpp] Error 2

  gmake[2]: *** [CMakeFiles/Makefile2:2679: CMakeFiles/fpp_locs.dir/all]
  Error 2

  gmake[1]: *** [CMakeFiles/Makefile2:2686: CMakeFiles/fpp_locs.dir/rule]
  Error 2

  gmake: *** [Makefile:498: fpp_locs] Error 2
Call Stack (most recent call first):
  fprime/cmake/sub-build/sub-build.cmake:65 (execute_process_or_fail)
  fprime/cmake/FPrime.cmake:152 (run_sub_build)
  fprime/cmake/FPrime.cmake:221 (fprime_initialize_build_system)
  CMakeLists.txt:13 (include)

[ERROR] CMake erred with return code 1. Partial build cache remains. Run purge to clean-up.

## How to Reproduce

I am using Ubuntu 22.04.3 LTS

1. python3 -m venv fprime-venv
2. . fprime-venv/bin/activate
3. pip install -U fprime-tools
4. fprime-util new --project
5. cd NewProject
6. fprime-util generate

LeStarch, 2024-01-28:
This can be caused by a bad package install.  Try this:  with the virtual environment sources, run the following:
 
pip install  -U --force-reinstall -r MyProject/fprime/requirements.txt

LeStarch, 2024-01-28:
This will force a reinstallation of the package and clean things up.

dury2379, 2024-01-28:
Command `pip install -U --force-reinstall -r MyProject/fprime/requirements.txt` fails with this error:
```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
fprime 1.5.4.dev0 requires Cheetah3==3.2.6.post2, but you have cheetah3 3.2.6.post1 which is incompatible.
fprime 1.5.4.dev0 requires lxml==4.6.3, but you have lxml 4.9.3 which is incompatible.
fprime 1.5.4.dev0 requires Markdown==3.3.4, but you have markdown 3.5 which is incompatible.
fprime 1.5.4.dev0 requires pytest==6.2.4, but you have pytest 7.4.3 which is incompatible.
```

thomas-bc, 2024-01-28:
@dury2379 it seems like you might be using an old version of Python and/or pip - the newest versions of F' requires Python 3.8 or above. If that's already what you're using, try upgrading pip first and then upgrade/force reinstall fprime-tools

dury2379, 2024-01-28:
My Python version is:
```
python3 --version
Python 3.10.12
```
pip version:
```
pip --version
pip 23.3.2 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)
```
Error is still there

thomas-bc, 2024-01-28:
@dury2379 hmm something is off then, the error message seems to indicate you've got fprime-tools==1.5.4, latest should be 3.4.1. Maybe start with fresh virtual environment, and see if 'pip install fprime-tools==3.4.1' will install the correct version.

dury2379, 2024-01-28:
I uninstalled fprime with this command:
`sudo pip uninstall fprime`
And then ran:
`pip install -U --force-reinstall -r fprime/requirements.txt `
With the idea in mind that dependency handler will get the right version. It seems it did because the last command ran without errors.
The original error happens when I run `fprime-util generate` still there.

JohanBertrand, 2024-01-28:
@dury2379 I could reproduce the issue but it came out with a different error; it seems like it is related to the `F'` parent folder

For some reasons the `'` disappears from the path while executing a python script. 

Try to rename the `F'` folder to `Fprime`, and that should solve the issue.

For reference, here is the error I got from my side:

```
CMake Error at fprime/cmake/utilities.cmake:586 (message):
-- Configuring incomplete, errors occurred!
  [sub-build] Failed to execute: info-cache/fpp_locs:

  /usr/bin/python3: can't open file
  '/workspaces/F/MyProject/fprime/cmake/target/tools/redirector.py
  /workspaces/F/MyProject/build-fprime-automatic-native/sub-build-info-cache/locs.fpp':
  [Errno 2] No such file or directory

  gmake[3]: *** [CMakeFiles/fpp_locs.dir/build.make:74:
  /workspaces/F'/MyProject/build-fprime-automatic-native/locs.fpp] Error 2

  gmake[2]: *** [CMakeFiles/Makefile2:1864: CMakeFiles/fpp_locs.dir/all]
  Error 2

  gmake[1]: *** [CMakeFiles/Makefile2:1871: CMakeFiles/fpp_locs.dir/rule]
  Error 2

  gmake: *** [Makefile:124: fpp_locs] Error 2
Call Stack (most recent call first):
  fprime/cmake/sub-build/sub-build.cmake:65 (execute_process_or_fail)
  fprime/cmake/FPrime.cmake:152 (run_sub_build)
  fprime/cmake/FPrime.cmake:221 (fprime_initialize_build_system)
  CMakeLists.txt:13 (include)
```

thomas-bc, 2024-01-29:
Also FYI, 

> pip --version
pip 23.3.2 from /usr/local/lib/python3.10/dist-packages/pip (python 3.10)

`/usr/local/lib/<...>` shows that this is not the pip that's within your virtual environment, but the system pip. Make sure you are in the correct virtual environment when you run all commands. Please see if you can follow https://nasa.github.io/fprime/INSTALL.html or if any item in the troubleshooting guide can help

dury2379, 2024-01-30:
Sorry for the radio silence. While trying to figure the issue out on my own, something went wrong and the window manager died. I reinstalled Ubuntu. The error does not happen on the fresh install. If someone down the line encounters the same error, no non-destructive solution has been found yet.

closed 2024-01-30 by dury2379