nasa/fprime #2505: Space In Project Description, when using Fprime-utils new package generates errors in fprime-utils generate

weaselslider, 2024-01-30:
| | |
|:---|:---|
|**_F´ Version_**| | Master branch, 3.3.2 & devel
|**_Affected Component_**|   |  prime-util
---
## Problem Description

Making a new project, using` fprime-util new --project ` creates errors.

```
(fprime-test) weaselslider@Weasel-Laptop:/mnt/c/Users/wease/Desktop/Comp4710/fprime-test-test/Hello World$ fprime-util generate
[INFO] Generating build directory at: /mnt/c/Users/wease/Desktop/Comp4710/fprime-test-test/Hello World/build-fprime-automatic-native
[INFO] Using toolchain file None for platform default
CMake Error: Could not find cmake module file: CMakeDetermineWorldCompiler.cmake
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_World_COMPILER_ENV_VAR
CMake Error: Error required internal CMake variable not set, cmake may not be built correctly.
Missing variable is:
CMAKE_World_COMPILER
CMake Error: Could not find cmake module file: /mnt/c/Users/wease/Desktop/Comp4710/fprime-test-test/Hello World/build-fprime-automatic-native/CMakeFiles/3.22.1/CMakeWorldCompiler.cmake
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
CMake Error at CMakeLists.txt:7 (project):
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
  No CMAKE_World_COMPILER could be found.

  Tell CMake where to find the compiler by setting the CMake cache entry
  CMAKE_World_COMPILER to the full path to the compiler, or to the compiler
  name if it is in the PATH.

CMake Error: Could not find cmake module file: CMakeWorldInformation.cmake
CMake Error: CMAKE_World_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!
See also "/mnt/c/Users/wease/Desktop/Comp4710/fprime-test-test/Hello World/build-fprime-automatic-native/CMakeFiles/CMakeOutput.log".
[ERROR] CMake erred with return code 1. Partial build cache remains. Run purge to clean-up.
```

The CMakeLists.txt is as follows
```
(fprime-test) weaselslider@Weasel-Laptop:/mnt/c/Users/wease/Desktop/Comp4710/fprime-test-test/Hello World$ cat CMakeLists.txt
####
# This sets up the build system for the 'Hello World' project, including
# components and deployments from project.cmake. In addition, it imports the core F Prime components.
####

cmake_minimum_required(VERSION 3.13)
project(Hello World C CXX)

###
# F' Core Setup
# This includes all of the F prime core components, and imports the make-system.
###
include("${CMAKE_CURRENT_LIST_DIR}/fprime/cmake/FPrime.cmake")
# NOTE: register custom targets between these two lines
include("${FPRIME_FRAMEWORK_PATH}/cmake/FPrime-Code.cmake")

# This includes project-wide objects
include("${CMAKE_CURRENT_LIST_DIR}/project.cmake")
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
(fprime-test) weaselslider@Weasel-Laptop:/mnt/c/Users/wease/Desktop/Comp4710/fprime-test-test/Hello World$ fprime-util version-check
usage: fprime-util [-h] {build,check,generate,purge,fpp-check,fpp-to-xml,visualize,impl,hash-to-file,info,new,format} ...
fprime-util: error: argument command: invalid choice: 'version-check' (choose from 'build', 'check', 'generate', 'purge', 'fpp-check', 'fpp-to-xml', 'visualize', 'impl', 'hash-to-file', 'info', 'new', 'format')
```

## How to Reproduce

1. fprime-util new --project
2. name the project something with a space in it (I used "Hello World")
3. `fprime-util generate` in the generated project

## Expected Behavior
fprime-util should either prevent the user from creating a project with a space character in the name, or it should format the CMakeLists.txt correctly, with quotes around the project name.

thomas-bc, 2024-01-30:
Somewhat a duplicate of https://github.com/nasa/fprime/issues/2322

Using spaces in Unix paths really isn't recommended. Is there a behavior you'd like to support here?
We should just have `fprime-util new` error out if a space is provided. I think this is implemented for new components, but hasn't been adapted to deployments and projects. This could be a good first PR to make if you'd like to!

thomas-bc, 2024-01-31:
Implemented the check for invalid characters

closed 2024-01-31 by thomas-bc