nasa/fprime #2508: `fprime-util build` fails with obscure error in autocoder if `fpp` does not include `queue size`

kbotteon, 2024-01-31:
| | |
|:---|:---|
|**_F´ Version_**| 3.4.0
|**_Affected Component_**|   `fprime-util build`
---
## Problem Description

If the `queue size` property of an Active Component is mistakenly omitted from the `fpp` file, the autocoder will happily proceed, and then the compile stage will fail with:

```
.../DeploymentAc.cpp: error: <component> is not a member of `<deployment>::QueueSizes’
```

It is a non-obvious that the `fpp` file is the source of the problem, with the only hint being that the error occurs in an autocoded file.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
fprime-util version-check
usage: fprime-util [-h]
                   {build,check,generate,purge,fpp-check,fpp-to-xml,visualize,impl,hash-to-file,info,new,format} ...
fprime-util: error: argument command: invalid choice: 'version-check' (choose from 'build', 'check', 'generate', 'purge', 'fpp-check', 'fpp-to-xml', 'visualize', 'impl', 'hash-to-file', 'info', 'new', 'format')
```

However, the `fprime-util -h` command reports `fprime-util (3.4.0)`

## How to Reproduce

1. Add an Active Component to a topology
2. Omit the `queue size` property
3. Run `fprime-util build`
4. Observe autocoding succeed and compiling fail

## Expected Behavior

`fprime-util` should complain about an invalid FPP file before the build process gets to the compile stage.

thomas-bc, 2024-01-31:
I think this is a check that should be performed as part of FPP, rather than through fprime-util and/or cmake. @bocchino do you have any input here?

bocchino, 2024-02-01:
This looks like a regression in FPP introduced in this commit:

https://github.com/fprime-community/fpp/commit/49df57f8304a713fc5f565ee6b08564ea2cf98d7

We removed the queue size check by accident when revising the stack size and priority checks.

bocchino, 2024-02-01:
The spec and User's Guide both say the right thing (queue size is required). We just need to undo the regressive change to the compiler.

closed 2024-02-01 by bocchino