nasa/fprime #2514: In the MathComponent Tutorial the MathSend unit test does not work

ljreder, 2024-02-02:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   MathSend unit test|
---
## Problem Description

The unit test in the `MathSender::testDoMath` method is failing at the assert
```
    // verify that the math operation port was invoked once
    ASSERT_from_mathOpOut_SIZE(1);
```
Note that MathSenderTesterHelper.cpp is being generated but not used.  When I attempt to add it to build I get errors and build fails.  @LeStarch said it should not be generated in a slack chat.  

Should it be set(UT_AUTO_HELPERS ON) or set(UT_AUTO_TEST_HELPERS ON) in the tutorial.  The first one works ok to build.

A description of the problem with sufficient detail to understand the issue.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
The version-check does not appear to be implemented????
```

## How to Reproduce

1.  Follow the steps in the MathSender unit test part of the MathComponent tutorial.  It should build ok.  
2.
3.

## Expected Behavior

Expected all tests to pass ok.

Here is the test log and test/ut files for reference and reproduction of problem
[LastTest.log](https://github.com/nasa/fprime/files/14136783/LastTest.log)

[MathSenderTester.cpp.txt](https://github.com/nasa/fprime/files/14136823/MathSenderTester.cpp.txt)
[MathSenderTester.hpp.txt](https://github.com/nasa/fprime/files/14136824/MathSenderTester.hpp.txt)
[MathSenderTestMain.cpp.txt](https://github.com/nasa/fprime/files/14136825/MathSenderTestMain.cpp.txt)

ljreder, 2024-02-06:
Thanks @thomas-bc I will have to take a closer look.  But I just do not get it.  I have been using code in the helper file for years and never thought it advanced.  At any rate it would be easier from a user point of view (e.g. my) if you guys just always generate *Helper.cpp files and build them.  The user should not need to use these at first but I think it would be more straight forward.  And then in tutorial let me them know it exists and is more advanced.  Just a thought.

thomas-bc, 2024-02-06:
Totally agree with you. It definitely gives more control, so for experienced users it may be preferable. I guess by "advanced" I just meant it abstracts more things away, so may be less overwhelming to new users.

thomas-bc, 2024-02-16:
@ljreder can you confirm that you're seeing the same issue that was just reported here: https://github.com/fprime-community/fprime-tutorial-math-component/issues/26 ?

thomas-bc, 2024-04-11:
Addressed by https://github.com/fprime-community/fpp/pull/401

ljreder, 2024-04-11:
I am a little confused by all these repo's but I just created the MathProject from scratch and built it all ok.  I added the unit test code for the MathSender and it fails again.  It looks all ok so I think some fpp did not get into the fprime yet or something.  I am showing it is using fprime version

HEAD detached at v3.4.3

Here is the test log file
 
[LastTest.log](https://github.com/nasa/fprime/files/14942774/LastTest.log)

Is there some simple way I can update the fprime submodule and rerun the build and test case to demonstrate that this thing does indeed work.

I have a fork of the fprime-community repo for the math tutorial but it would not build with the virtual environment from my MathProject so I just did not try it more.

Any suggestions to verify that this does indeed work ok.

thomas-bc, 2024-04-22:
@ljreder apologies for the late response. This has been fixed and released in FPP, but F´ itself hasn't been re-released yet so that may be why you are encountering issues.

What you can try is in your virtual environment
```bash
pip install -U fprime-fpp==2.1.0a7
```
Make sure it installed properly
```bash
pip freeze | grep fprime-fpp-*
```
should show the correct version.

Then you should purge your build cache `fprime-util purge` and run the `fprime-util impl` step again, as well as the steps of the tutorial that are relevant after that.

thomas-bc, 2024-10-16:
Fix by v3.5.0 release

closed 2024-10-16 by thomas-bc