nasa/fprime #2515: fprime-utils low hanging fruit.

ljreder, 2024-02-02:
Some simple features to fprime-utils are needed

1. Report version information with a simple version option.
2. Create the test/ut subdirectories when doing an impl --ut.
3. Auto move all generated files to test/ut when doing an impl --ut.

As I worked through the MathComponent tutorial I was surprised these things not present.

LeStarch, 2024-02-14:
1 :--help returns tool version (consistent with fpp), and we also have version_check to display all version information.
2, 3 we agree with, but will hold off until FPP generates UT source files with a `.template` suffix as otherwise overwrites are a problem!

AlesKus, 2024-12-11:
@LeStarch, as far as I see the [fpp-to-cpp](https://github.com/nasa/fpp/tree/main/compiler/lib/src/main/scala/codegen/CppWriter) still generates UT sources without `.template`, And there is no any [request ](https://github.com/nasa/fpp/issues)to implement it.
What if we force renaming  of the UT files? Something like this:
```python
# filename is a pathlib.Path object 
template_suffix = ".template"
if template_suffix not in filename.suffixes:
    filename = filename.with_suffix(template_suffix + filename.suffix) 
```

LeStarch, 2024-12-12:
@AlesKus yes, theoretically we could perform this change int the `fprime-util` layer.  If doing this, we likely should also make the `test/ut` directory and place files in it.

LeStarch, 2025-01-15:
This is in review and is pending merge on PRs to the tutorials!

closed 2025-02-11 by thomas-bc