nasa/fprime #2516: Flag to stop printing log messages to STDOUT

rrieber, 2024-02-05:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

The current FPrime behavior for log messages is to:
1. Send them to downlink,
2. Record them to the a file, and
3. print them to STDOUT.

In the event an operator wants to see STDOUT, one must download the STDOUT log file, which also includes the log messages. Thusly, those are received in duplicate, or in triplicate if the recorded stream is also telemetered. This reduces the signal to noise ratio of STDOUT and makes them unnecessarily large. Such a size may motivate an operator to triage STDOUT and skip downlinking it.

I propose that FPrime put in some sort of flag (a parameter, command, or a variable of sorts, enable or disable printing the messages to STDOUT.

## Rationale
This feature would preserve the logs to stdout for debugging during development, yet conserve bandwidth during operations.

Joshua-Anderson, 2024-02-05:
Not sure this needs a F' feature. Depending on your init program, there are various ways to stop fprime to logging to STDOUT, either a good old fashioned `> /dev/null` or more complicated options with systemd

LeStarch, 2024-02-05:
What log messages are you referring to here?  Is it `Fw::Logger` or `Fw::Log` aka Events?

`Fw::Logger` goes to a project-defined handler which is typically `stdout`.

Events are downlinked, and do go to `stdout` when the `TextLog` ports are hooked up.  i.e.

```
    text event connections instance textLogger
```

Projects that wish just to see stdout and not `events`  simply need to delete that line from the topology.

rrieber, 2024-02-06:
@LeStarch @Joshua-Anderson I think I wrote this ticket in haste. The log messages I was referring to are the ones comparable to EVRs. (`Fw:Log`?) And I recently learned about the project-defined handler that you also mention. I'm going to close this ticket and take this one on the chin.

closed 2024-02-06 by rrieber