nasa/fprime #2520: [GDS] StandardPipeline not correctly disconnected: `50050: Address already in use`

thomas-bc, 2024-02-07:
### Discussed in https://github.com/nasa/fprime/discussions/2519

<div type='discussions-op-text'>

<sup>Originally posted by **SMorettini** February  7, 2024</sup>
Since I started using fprime-gds, I've encountered the following error when starting it multiple times in a row:

```
[ERROR] Error with address/port '0.0.0.0:50050' : [Errno 98] Address already in use
```

This error doesn't occur every time but it seems to happen at least one out of ten times. My usual workaround is to change the `--tts-port`.

Upon investigation today, I found out what I suspected to be the problem. In `flask/app.py`, a pipeline is created but the `disconnect` method is never called. I attempted to add the call to `disconnect` but couldn't find a way to do so.

The pipeline is created only once when starting `fprime-gds`. Any request uses the same pipeline object. As far as I understand, Flask doesn't support the design of using the same object across all API requests. According to Flask's suggested design, each API function should create a new connection to the `TCPServer`. Therefore, Flask doesn't provide a way to register a function to clean up or teardown a global object.

I tried to register a `signal_handler` but it seems that Flask gives very little time for any signal handling, forcing the server to close too quickly and not allowing the `disconnect` method to complete execution.

Has anyone else encountered the `Address already in use` error? How did you solve it? Any ideas on how we could change fprime-gds to close more cleanly?</div>

zimri-leisher, 2024-02-07:
Also experiencing this.

thomas-bc, 2024-03-18:
The `--zmq` flag removes the need of a middleware application occupying port 50050, using ZMQ instead of a TCP server being the middleman. We will probably make `--zmq` flag the default, removing this issue altogether. 

Until a GDS release comes out, `fprime-gds --zmq --zmq-server` should work.

LeStarch, 2024-04-24:
Make zmq the default.

closed 2024-05-06 by thomas-bc

keck-in-space, 2025-11-28:
I am still experiencing this in F Prime 4.0.0 with ZMQ.

` Error with address/port of '0.0.0.0:50000' : [Errno 98] Address already in use`