nasa/fprime #2533: Build fails after creating a new deployment

ryan-halbrook, 2024-02-18:
| | |
|:---|:---|
|**_F´ Version_**|3.4.3 |
|**_Affected Component_**|  |
---
## Problem Description

I ran into this issue when going through the HelloWorld tutorial. After creating a deployment, the build fails:

```
[sub-build] Failed to execute: info-cache/fpp_locs:

  fpp-locate-defs

  /home/ryan/fprime/HelloWorld/MyDeployment/Top/topology.fpp:85.39

        comStub.drvDataOut -> comDriver.send
                                        ^

  error: identifier expected

  gmake[4]: *** [CMakeFiles/fpp_locs.dir/build.make:74: ../locs.fpp] Error 1

  gmake[3]: *** [CMakeFiles/Makefile2:1920: CMakeFiles/fpp_locs.dir/all]
  Error 2

  gmake[2]: *** [CMakeFiles/Makefile2:1927: CMakeFiles/fpp_locs.dir/rule]
  Error 2

  gmake[1]: *** [Makefile:124: fpp_locs] Error 2
```

Reverting the changes in requirements.txt from commit c6f8e7b resolved the issue for me.

## Context / Environment

```
Operating System: Linux
CPU Architecture: aarch64
Platform: Linux-5.15.0-1046-raspi-aarch64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 24.0
Pip packages:
    fprime-tools==3.4.3
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3
```

## How to Reproduce

1. Run `fprime-util new --deployment` and select defaults
2. Run `fprime-util build -j4`
3. Build fails with above error.

## Expected Behavior

Build passes.

thomas-bc, 2024-02-18:
Ah, that is a bug in the new release indeed... we'll ship a new release ASAP. In the meantime the fix is to change in the offending line:

```
comStub.drvDataOut -> comDriver.$send
```

Adding the $ (escape character in FPP) in front of send, as "send" is now a reserved keyword in FPP.

thomas-bc, 2024-02-20:
Fixed in v3.4.3 - thank you for reporting this!

closed 2024-02-20 by thomas-bc