nasa/fprime #2550: Port with string arg fails if in namespace, works otherwise

zimri-leisher, 2024-02-29:
| | |
|:---|:---|
|fprime-fpl-convert-xml |1.0.0|
|fprime-fpl-extract-xml |1.0.0|
|fprime-fpl-layout      |1.0.0|
|fprime-fpl-write-pic   |1.0.0|
|fprime-fpp             |1.3.0|
|fprime-gds             |3.3.0|
|fprime-tools           |3.4.4|
|fprime-visual          |1.0.0|
---
## Problem Description

I have a port:
```
module ports {
    port setIntVar(
        name: string
        value: I64
    )
}
```
I use it in a component:
```
        async input port setIntVar: ports.setIntVar
```
I get this error:
```
.../ComponentAc.hpp:408:17: error: ‘setIntVarPortStrings’ does not name a type
  408 |           const setIntVarPortStrings::StringSize80& name,
      |                 ^~~~~~~~~~~~~~~~~~~~
```

If I use any type other than `string`, it works (even a type with a string inside of it). If I specify the size of the string, it still doesn't work.

When I move the port out of the module like so:

```
port setIntVar(
    name: string
    value: I64
)
module ports {
    ...
}
```
And use it without the prefix like so:
```
        async input port setIntVar: setIntVar
```

The component compiles.
My guess is this is a bug in the autocoder, forgetting to add a `ports::` prefix to the `setIntVarPortStrings::StringSize80` type.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.29
Python version: 3.8.10
CMake version: 3.16.3
Pip version: 21.2.4
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.3.0
[WARNING] The 'fprime-fpp-to-xml' distribution was not found and is required by the application
[WARNING] The 'fprime-fpp-to-json' distribution was not found and is required by the application
[WARNING] The 'fprime-fpp-to-cpp' distribution was not found and is required by the application
[WARNING] The 'fprime-fpp-syntax' distribution was not found and is required by the application
[WARNING] The 'fprime-fpp-locate-uses' distribution was not found and is required by the application
[WARNING] The 'fprime-fpp-locate-defs' distribution was not found and is required by the application
[WARNING] The 'fprime-fpp-from-xml' distribution was not found and is required by the application
[WARNING] The 'fprime-fpp-format' distribution was not found and is required by the application
[WARNING] The 'fprime-fpp-filenames' distribution was not found and is required by the application
[WARNING] The 'fprime-fpp-depend' distribution was not found and is required by the application
[WARNING] The 'fprime-fpp-check' distribution was not found and is required by the application
```

LeStarch, 2024-02-29:
@bocchino we just talked about port strings.  Did you have any insight here?

bocchino, 2024-02-29:
This issue should be fixed as of this PR: https://github.com/fprime-community/fpp/pull/321. It looks like the version of FPP being used (1.3.0) predates this PR.

zimri-leisher, 2024-02-29:
Thank you! I will patch our fork with this and let you know if it fixes the problem.

zimri-leisher, 2024-03-04:
I'm closing this because this issue is apparently already solved. I don't think we're going to update our internal fork for this fix, we will just use a new type which only has a string inside of it.

closed 2024-03-04 by zimri-leisher