nasa/fprime #2551: Add tests to `fprime-bootstrap` repo

thomas-bc, 2024-02-29:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**| `fprime-bootstrap` |
---
## Feature Description

We have a new tool, https://github.com/fprime-community/fprime-bootstrap . In my haste to make it available, I skipped writing simple tests to make sure things don't break down the line. It would be nice to write some simple tests that run the module and check that files are generated. Two things I'd like to see tested is:
- no file in the generated directory are named `*-template`
- no file in the generated directory contain the string `{{FPRIME_PROJECT_NAME}}`

As either of these would indicate something happened during the generation.

Bonus: these tests should be ran in CI, just like we do for example here https://github.com/fprime-community/fprime-tools/blob/devel/.github/workflows/fprime-tools-ci.yml (probably can reuse the exact same workflow definition)

thomas-bc, 2024-02-29:
If anyone would like to work on this, please open a PR and tag me for review!

matt392code, 2024-03-01:
Would I be able to take a crack at it?
#thomas-bc

thomas-bc, 2024-03-01:
Yes of course :)

matt392code, 2024-03-01:
[image]

matt392code, 2024-03-02:
Thank you for the extra information.
I will have to spend some time reading up on the F Prime documentation to make sure I am doing it correctly.  Will use Debian Linux.

thomas-bc, 2024-03-02:
Using F´bootstrap is explained here: https://fprime-community.github.io/fprime-tutorial-hello-world/docs/NewProject.html

This should be enough to understand what it's supposed to do.

matt392code, 2024-03-03:
Set up F Prime on a Debian Linux emulation using VirtualBox and working my way through the tutorials.  Having some issues building.  Will rest and take another look at it in a day or so.

matt392code, 2024-03-05:
Still working on  it.  Researching a CMake error.

matt392code, 2024-03-07:
Making progress.

matt392code, 2024-03-14:
Moving forward.  Sorry for the delay.

thomas-bc, 2024-03-14:
No problem at all, there's no rush. Let me know if you have any questions

matt392code, 2024-03-14:
Finished HelloWorld tutorial:
[image]

matt392code, 2024-03-14:
[image]

matt392code, 2024-03-17:
Working on Math tutorial

matt392code, 2025-01-11:
Proposed tests that verify the project generation and check for template artifacts:
[fprime-bootstrap-tests.py.txt](https://github.com/user-attachments/files/18383917/fprime-bootstrap-tests.py.txt)
Proposed GitHub Actions workflow file to run these tests in CI:
[github-workflow.txt](https://github.com/user-attachments/files/18383919/github-workflow.txt)
Two key components:

1. A test suite (`fprime-bootstrap-tests`) that verifies:
   - No files with `-template` suffix exist in generated projects
   - No template strings (`{{FPRIME_PROJECT_NAME}}`) remain in generated files
   - Basic project structure is created correctly
   
2. A GitHub Actions workflow configuration that:
   - Runs on multiple Python versions (3.8-3.11)
   - Installs the package in development mode
   - Runs the test suite
   - Performs an actual project generation test

To use these tests:
1. Place the test file in a `tests` directory in your repository (e.g., `tests/test_bootstrap.py`)
2. Place the workflow file in `.github/workflows/fprime-bootstrap-ci.yml`
3. Update your `setup.py` or `pyproject.toml` to include test dependencies

The tests follow a similar pattern to fprime-tools CI while focusing on the specific requirements for bootstrap. They verify both the absence of template artifacts and successful project generation.

thomas-bc, 2025-01-11:
@matt392code these AI-generated comments are not very helpful in most cases and spam our inboxes. May I ask that you refrain from posting these? Or maybe review their content and ensure they are accurate before posting?
Thank you very much for trying to help the community though!

matt392code, 2025-01-11:
Ok, sorry about that.
Thought it would be helpful since they were older tickets.
Matt

&mdash;

thomas-bc, 2025-01-11:
No problem at all, thank you for contributing! 
It's just that the volume of content that is generated is hard for our team to sort through. If there are tickets where we would like some input, I will make sure to let you know.

Thanks!

matt392code, 2025-01-11:
Ok, sounds good.
Sorry for the volume - my thought was that I could use AI to do the grunt work for the senior engineers, then the people with the high level skills/experience [well above my level] could make a few modifications then test the code before committing it to the repository.

rmzmrnn, 2025-06-19:
Hey @thomas-bc 👋

Just saw this issue and I'd love to give it a shot if no one’s picked it up yet. Let me know if that's okay!