nasa/fprime #2558: Single quote in project parent folder causes installation errors

TSKoduru, 2024-03-04:
| | |
|:---|:---|
|**_F´ Version_**| v3.4.3 |
|**_Affected Component_**|  Installation/Project Setup |
---
## Problem Description

If you name the parent directory of your project such that it has a single quote in it, you'll run into an error when you try to run 'fprime-util generate' during basic project setup. 

## Context / Environment

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 24.0
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3
```

## How to Reproduce

1.  Create a parent directory containing a single quote (For instance, "F'")
2.  Follow the basic project setup guide located here: https://fprime-community.github.io/fprime-tutorial-hello-world/docs/NewProject.html 

## Expected Behavior

Ideally, you'd have a project set up properly in your virtual environment.

thomas-bc, 2024-03-04:
Note: having a space in the parent directory also causes issues.

TSKoduru, 2024-03-04:
Yeah, that's honestly going to be the best bet. At the very least, we could have a warning on the NewProject website letting you know about the potential issues with certain project names.

thomas-bc, 2024-03-06:
To be fixed by https://github.com/nasa/fprime/issues/2570

closed 2024-03-07 by thomas-bc