nasa/fprime #2563: Cross Compile defaults to an incorrect toolchain

NathanJSnyder, 2024-03-05:
| | |
|:---|:---|
|**_F´ Version_**| v3.4.3| 
|**_Affected Component_**| n/a |
---
## Problem Description

F' is failing to find the cross compile toolchain and defaults to /usr/bin/cc. This should show an error.

## Context / Environment

Executed `fprime-util version-check`

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.4.0-172-generic-x86_64-with-glibc2.29
Python version: 3.8.10
CMake version: 3.16.3
Pip version: 24.0
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3
```

Executed `fprime-util generate aarch64-linux -DCMAKE_DEBUG_OUTPUT=ON`

```
-- [arm-linux] Assembler:    CMAKE_ASM_COMPILER-NOTFOUND
-- [arm-linux] C   Compiler: CMAKE_C_COMPILER-NOTFOUND
-- [arm-linux] CXX Compiler: CMAKE_CXX_COMPILER-NOTFOUND
-- [arm-linux] Assembler:    CMAKE_ASM_COMPILER-NOTFOUND
-- [arm-linux] C   Compiler: CMAKE_C_COMPILER-NOTFOUND
-- [arm-linux] CXX Compiler: CMAKE_CXX_COMPILER-NOTFOUND
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
```

## How to Reproduce

1. Attempt to cross compile a project on this platform. (led-blinker was used in this case)

roemvaar, 2024-03-12:
Is this a problem that needs to be fixed and what would be the "definition of done"? Just failing the compilation and displaying that no cross compiler was found? 

If that's the case, I can tackle it

NathanJSnyder, 2024-03-12:
It does need to be fixed and that is the expected solution. Thank you for looking into it!

roemvaar, 2024-03-18:
The build is unsuccessful using a newer version of CMake. F' version is the same as the problem report.

Executed `fprime-util version-check`

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-6.5.0-25-generic-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 24.0
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3
```

Executed `fprime-util generate aarch64-linux -DCMAKE_DEBUG_OUTPUT=ON`

```
[INFO] Generating build directory at: /local/directory/led-blinker/build-fprime-automatic-aarch64-linux
[INFO] Using toolchain file /local/directory/led-blinker/fprime/cmake/toolchain/aarch64-linux.cmake for platform aarch64-linux
CMake Error at fprime/cmake/toolchain/helpers/arm-linux-base.cmake:20 (find_program):
-- Configuring incomplete, errors occurred!
  Could not find CMAKE_CXX_COMPILER using the following names:
  aarch64-linux-gnu-g++, aarch64-none-linux-gnu-g++
Call Stack (most recent call first):
  fprime/cmake/toolchain/aarch64-linux.cmake:18 (include)
  /usr/share/cmake-3.22/Modules/CMakeDetermineSystem.cmake:124 (include)
  CMakeLists.txt:7 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
[ERROR] CMake erred with return code 1. Partial build cache remains. Run purge to clean-up.
```

I'll try to use the same CMake version as the original bug report and see if I can reproduce the bug. Let me know if you have any thoughts.

thomas-bc, 2024-03-18:
```CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.```

That's a really odd error that I've never seen before, does `make --version` work?
What about `fprime-util generate` without the platform?

It may be weird artifacts of the earlier 
```
  Could not find CMAKE_CXX_COMPILER using the following names:
  aarch64-linux-gnu-g++, aarch64-none-linux-gnu-g++
```

This seems to indicate your cross compilers are not installed properly. Which also seems to be why the issue appeared for OP (see `-- [arm-linux] CXX   Compiler: CMAKE_CXX_COMPILER-NOTFOUND` in the original bug report)

LeStarch, 2024-03-19:
Also, has `make` been installed on your linux system?  You will need to install the cross-compilers too.

roemvaar, 2024-03-20:
@thomas-bc I specifically didn't follow the "cross-compilation setup steps" to try and replicate the bug mentioned on the original post. 

> That's a really odd error that I've never seen before, does make --version work?

```
GNU Make 4.3
Built for x86_64-pc-linux-gnu
```

> What about fprime-util generate without the platform?

This worked just fine.

@LeStarch `make` and all build tools, except for the cross-compiler, are installed on my setup. 

I know my setup is working correctly because I was able to complete the `HelloWorld` tutorial. Now, I'm following the `LedBlinker` tutorial to turn on a LED on a BeagleBoneBlack. I know that if I install the xcompile tools and pass them as arguments to `make` the code will compile just fine, but I wanted to check what happened if I don't specify the directory of the aarch64 xcompiler. I was wondering if it would default to the native compiler for x86 (as mentioned on the bug report).

Any thoughts on what to do next to complete this task?

thomas-bc, 2024-03-20:
Ohhh ok I understand. Then yes in my opinion it would make sense to error out in the `generate` step if xcompilers are not found, instead of defaulting back to the host's compilers. @LeStarch thoughts?

LeStarch, 2024-03-20:
You need to set the path to the cross-compiler tools.  Otherwise this setup won't work.  To run the native (x86_64) compiler you may use:

```
fprime-util generate native
fprime-util build native
```

LeStarch, 2024-03-20:
Ohhhhhh.....I lost track.  This issue is in the original build, the CC compiler was used despite failing to find the ARM compiler.  We need to error out.

thomas-bc, 2024-03-20:
@roemvaar if you would like to tackle this, it would be a really good first issue to get familiar with our CMake system. Let us know!

roemvaar, 2024-03-21:
Absolutely, I want to tackle this :)

LeStarch, 2025-01-15:
@roemvaar are you still working on this one?  Here are the steps:

0. Ensure you can reproduce the error
1. Add a check for tools being found after the `find_program` calls
2. message `FATAL_ERROR` if not found if not found
3. Test with tools not installed (using reproduction in step 0)
4. Test with tools installed (nominal case).

e.g. On this line https://github.com/nasa/fprime/blob/5a7452a474fb3c78147b2fad0936d317dd8e9da2/cmake/toolchain/helpers/arm-linux-base.cmake#L25 add something like:

```
if (NOT CMAKE_C_COMPILER OR NOT CMAKE_CXX_COMPILER OR ...)
    message(FATAL_ERROR "Failed to find Arm cross-compile toolchain")
endif()
```

roemvaar, 2025-01-16:
@LeStarch I got busy finishing my master's, but now I have time to tackle this. I will review the steps you proposed and provide an update by the end of the week.