nasa/fprime #2566: Force Git Clone To Select Unix Line Endings

LeStarch, 2024-03-06:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

fprime-bootstrap clones forums without specifying Unix line endings.  In some cases this can cause files to checkout with Windows line endings that then breaks tools.

We should supply get with designated flags to use Unix line ending.

## Context / Environment

WSL

thomas-bc, 2024-03-07:
@LeStarch see `core.autocrlf` here: https://git-scm.com/book/en/v2/Customizing-Git-Git-Configuration

It seems `input` is the config we want, but I am a little wary about setting this option for every system when really just a limited set should need it... do you have any experience with this? Can you confirm `input` is what we want to set?

LeStarch, 2024-03-07:
Perhaps the best approach is to add a note to the trouble shooting guide in install for WSL systems and collect more data before fixing this?  If so, mark this as will not fix.

thomas-bc, 2024-03-07:
Closing as wontfix, as it may introduce more issues than it will fix.

closed 2024-03-07 by thomas-bc