nasa/fprime #2573: Google Search Shows Old Versions

csmith608, 2024-03-07:
| | |
|:---|:---|
|**_F´ Version_**| 3.4.3 |
|**_Affected Component_**| Documentation  |
---
## Feature Description

Add a theme to outdated fprime documentation indicating that it is outdated and has been superseded by a newer version. 

## Rationale

To quickly find a part of an fprime tutorial to reference I'll use good to search. For example, I'll google "math component fprime" and the first hit is the tutorial [from fprime 2.0.0](https://nasa.github.io/fprime/v2.0.0/Tutorials/MathComponent/Tutorial.html). The second hit is from fprime 3.1.0. 
Adding a header would reduce confusion and is standard in some other repositories.

thomas-bc, 2024-03-11:
Changes probably need to be done in that branch here if what is requested is for the older versions of documentation: https://github.com/nasa/fprime/tree/docs/auto-documentation/docs

Not too sure what the best approach is to handle those kind of things - if anyone from the community wants to jump in and contribute for this, would be greatly appreciated!

LeStarch, 2024-03-12:
We should stamp old versions with a tag saying:

"Latest documentation available here: link to devel"

LeStarch, 2024-04-01:
One more to look into after the webwork!

rithvikrajraapeti, 2024-11-27:
So which file should i work on this issue

thomas-bc, 2024-12-06:
Hopefully the new website will get picked up by google search better. Versioning is now all handled within the F´ website. There's not much we can do about google search results AFAIK.

closed 2024-12-06 by thomas-bc