nasa/fprime #2578: Create F´ Interfaces From Existing Concepts

LeStarch, 2024-03-12:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Many F´ concepts can be reworked as F´ interfaces. These are specialized `.fppi` files that define a set of ports.  An example would be: https://github.com/nasa/fprime/blob/devel/Drv/Interfaces/ByteStreamDriverInterface.fppi.

Interfaces are expected to live in a folder `Interfaces` under a given package (e.g. `Drv/Interfaces` or `Svc/Sinterfaces`).

> New interfaces should use the filename pattern `*.interface.fppi` and not just `.fppi`. This is a change since the example was created.

Here is a list of interfaces that should be created:
- [ ] `Svc::ComInterface`:
- [ ] Event Ports:
- [ ] Telemetry Ports:
- [ ] Command Ports:
- [ ] I2C Driver:
- [ ] GPIO Driver:
- [ ] SPI Driver:
- [ ] Block Driver:
- [x] UART and TCP Driver: https://github.com/nasa/fprime/issues/2255
- [x] Time: https://github.com/nasa/fprime/issues/2256

LeStarch, 2025-05-09:
@kevin-f-ortega is this done?  The two outstanding bits were already implemented.

kevin-f-ortega, 2025-05-09:
@LeStarch, yes this is done now

closed 2025-05-10 by thomas-bc