nasa/fprime #2579: Rename Time Interface File

LeStarch, 2024-03-12:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

`Svc/Interfaces/Time.fppi` should be renamed to `Svc/Interfaces/TimeInterface.fppi` to match modern standards.

rldettloff, 2024-04-09:
I'm working on this!

LeStarch, 2024-05-10:
@rldettloff are you still working on this?

CombustableLem0n, 2024-05-13:
I added a pull request addressing this issue. I'm new to submitting pull requests but I think this should help you find my pull request. (gh pr checkout 2728)

CombustableLem0n, 2024-05-14:
I'm confused why some test checks failed first time on my pull request, but the same checks passed second with identical code. Any explanations?

thomas-bc, 2024-05-14:
@CombustableLem0n thank you for closing this out!

closed 2024-05-14 by thomas-bc