nasa/fprime #2582: Create Interface from LinuxSpiDriver

LeStarch, 2024-03-12:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Make an interface out of Drv/LinuxSpiDriver protocol. See #2255 for an example interface change set.

The following three ports should be pulled out of `Drv::LinuxSpiDriver` into `Svc/Interfaces/SpiInterface.fppi` such that other Spi implementations can use it.

https://github.com/nasa/fprime/blob/a73bff28617b96736c0e3e222e4c6e1dea1c6f5e/Drv/LinuxSpiDriver/LinuxSpiDriver.fpp#L9

Note: annotations should be added to explain what each port does.

menaman123, 2024-06-04:
Hello, I was wondering if there is documentation on what this port does. Im assuming it reads and writes?

menaman123, 2024-06-04:
Would the description be like this:
@ Port to perform synchronous read and write operations over the SPI bus

LeStarch, 2024-06-04:
I would say "Port to perform a synchronous read/write operation over the SPI bus"...but you basically have it.  The nuance I am trying to capture here is that SPI calls are both read and write at the same time.

menaman123, 2024-06-05:
Follow up question, do i need to add anything to the cmake file so that there wont be any sort of failures?

closed 2025-05-09 by LeStarch