nasa/fprime #2602: fprime-util generate fails with cmake 3.16.3

JohanBertrand, 2024-03-19:
| | |
|:---|:---|
|**_F´ Version_**| development |
|**_Affected Component_**| cmake |
---
## Problem Description

Fprime currently fails to generate with cmake version 3.16.3

It seems like it has been introduced with https://github.com/nasa/fprime/pull/2502

An option might be to change the minimum cmake version to a higher version, to revert the changes from the mentioned pull request or adapt the cmake file to be compatible with older versions.

```
root@9712ae2c742b:/home/fprime# fprime-util generate
[WARNING] /home/fprime/settings.ini does not exist
[INFO] Generating build directory at: /home/fprime/build-fprime-automatic-native
[INFO] Using toolchain file None for platform default
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [WARNING] /home/fprime/settings.ini does not exist
[WARNING] /home/fprime/settings.ini does not exist
FPRIME_FRAMEWORK_PATH=/home/fprime
-- [python3] python3 found at: /usr/bin/python3
-- [fpp-tools] fpp-depend found at: /usr/local/bin/fpp-depend
-- [fpp-tools] fpp-to-xml found at: /usr/local/bin/fpp-to-xml
-- [fpp-tools] fpp-to-cpp found at: /usr/local/bin/fpp-to-cpp
-- [fpp-tools] fpp-locate-defs found at: /usr/local/bin/fpp-locate-defs
-- [FPRIME] Module locations: /home/fprime
-- [FPRIME] Configuration module: /home/fprime/config
-- [FPRIME] Installation directory: /home/fprime/build-artifacts
-- Target build toolchain/platform: Linux/Linux
-- Including /home/fprime/cmake/platform/Linux.cmake
-- Requiring thread library
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- [sub-build] Performing sub-build: info-cache
-- [sub-build] Generating: info-cache with target/fpp_locs;target/fpp_depend
CMake Error at cmake/utilities.cmake:586 (message):
  [sub-build] Failed to generate: info-cache:

  CMake Error at cmake/utilities.cmake:604 (get_property):

    INTERFACE_LIBRARY targets may only have whitelisted properties.  The
    property "IMPLEMENTORS" is not allowed.

  Call Stack (most recent call first):

    cmake/API.cmake:619 (append_list_property)
    Os/Stub/CMakeLists.txt:14 (register_fprime_implementation)

  

  

  CMake Error at cmake/FPrime.cmake:196 (add_subdirectory):

    INTERFACE_LIBRARY targets may only have whitelisted properties.  The
    property "IMPLEMENTORS" is not allowed.

  Call Stack (most recent call first):

    cmake/FPrime-Code.cmake:15 (fprime_setup_included_code)
    CMakeLists.txt:67 (include)

  

  

  CMake Error at cmake/utilities.cmake:604 (get_property):

    INTERFACE_LIBRARY targets may only have whitelisted properties.  The
    property "IMPLEMENTORS" is not allowed.

  Call Stack (most recent call first):

    cmake/API.cmake:619 (append_list_property)
    Os/Posix/CMakeLists.txt:19 (register_fprime_implementation)

-- Configuring incomplete, errors occurred!
  

  

  CMake Error at cmake/FPrime.cmake:196 (add_subdirectory):

    INTERFACE_LIBRARY targets may only have whitelisted properties.  The
    property "IMPLEMENTORS" is not allowed.

  Call Stack (most recent call first):

    cmake/FPrime-Code.cmake:15 (fprime_setup_included_code)
    CMakeLists.txt:67 (include)

  

  

  CMake Error at cmake/utilities.cmake:604 (get_property):

    INTERFACE_LIBRARY targets may only have whitelisted properties.  The
    property "REQUESTERS" is not allowed.

  Call Stack (most recent call first):

    cmake/API.cmake:601 (append_list_property)
    Os/CMakeLists.txt:103 (require_fprime_implementation)

  

  

  CMake Error at cmake/FPrime.cmake:196 (add_subdirectory):

    INTERFACE_LIBRARY targets may only have whitelisted properties.  The
    property "REQUESTERS" is not allowed.

  Call Stack (most recent call first):

    cmake/FPrime-Code.cmake:15 (fprime_setup_included_code)
    CMakeLists.txt:67 (include)
See also "/home/fprime/build-fprime-automatic-native/CMakeFiles/CMakeOutput.log".
Call Stack (most recent call first):
See also "/home/fprime/build-fprime-automatic-native/CMakeFiles/CMakeError.log".
  cmake/sub-build/sub-build.cmake:39 (execute_process_or_fail)
  cmake/FPrime.cmake:152 (run_sub_build)
  cmake/FPrime.cmake:221 (fprime_initialize_build_system)
  CMakeLists.txt:12 (include)

[ERROR] CMake erred with return code 1. Partial build cache remains. Run purge to clean-up.
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.10.102.1-microsoft-standard-WSL2-x86_64-with-glibc2.29
Python version: 3.8.10
CMake version: 3.16.3
Pip version: 20.0.2
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a5
```

## How to Reproduce

1. Create a docker image on Ubuntu 20.04
2. Install needed packages, including cmake (base version is 3.16.3)
3. Run `fprime-util generate`

## Expected Behavior

The project should be generated properly

LeStarch, 2024-03-25:
Looks like it is an issue with versions of CMake before `3.18`.  These don't seem to support general properties on interface libraries.

For now, I suggest the work-around of upgrading CMake.   I will take a look, and see if we can bring back backwards compatibility.

LeStarch, 2024-03-25:
@thomas-bc we should probably have a CI test that runs the oldest supported CMake and make sure we do not have more issues of this sort.

PR #2617 (2024-03-26): Fixes #2602; bumps required version to 3.16
| | |
|:---|:---|
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Fixes #2602.  Brings "required CMake version" statements inline with INSTALL.md.

## Rationale

Small bug, no implementation downsides.  Makes CMake consistent with INSTALL.

## Future Work

Make CMake CI UTs use CMake 3.16.
commit 9f5b3645 (2024-03-26): Fixes #2602; bumps required version to 3.16 (#2617)
closed 2024-03-26 by LeStarch