nasa/fprime #2604: Split System Resources Into Version and System Reources Component

LeStarch, 2024-03-19:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

The System Resources component internalizes version telemetry, command, and events.  We should build a new version component that does the following:

1. Version events (EVRs) at start-up and upon command
2. Version telemetry channels (EHA)
3. Versions to track:
  1. Fprime framework version (git version of F´ repo)
  2. Project version (git version of project repo)
  3. Library versions (git version and name of each fprime-library repo)

thomas-bc, 2024-03-22:
Potentially in scope: https://github.com/nasa/fprime/issues/1837

Shivaly-Reddy, 2024-04-24:
@LeStarch : How do we track version at start-up? What part of the logic kicks of startup from version component?

LeStarch, 2024-04-24:
@Shivaly-Reddy F´ Active components allow you to override the `preamble` function to do something right when their thread starts-up.  This component is active, yes?

An example is:

https://github.com/nasa/fprime/blob/7724562530294a91c0bcf347e825220eb6feac03/Svc/ActiveRateGroup/ActiveRateGroup.hpp#L109

https://github.com/nasa/fprime/blob/7724562530294a91c0bcf347e825220eb6feac03/Svc/ActiveRateGroup/ActiveRateGroup.cpp#L55-L57

Note: the task lifecycle looks like this:

1. Call preamble
2. Loop on "dispatch message"

Thus the preamble is called once, and then all the messages (commands and port invocations) are dispatched for the duration.

Shivaly-Reddy, 2024-04-24:
Thanks @LeStarch. I was looking at FileDownlink preamble calls and was wondering about the component being active. It is currently a passive component, just because SystemResources was and because this is derived from it. I can switch it to an active component but need to fully understand the ramifications of an active component (other than needing a thread and a queue)

closed 2024-08-21 by thomas-bc