nasa/fprime #2612: Add Reusable FIT Workflow

LeStarch, 2024-03-25:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Currently, there is no reusable Github workflow step for FIT testing as [there is for building/UT](https://github.com/nasa/fprime/blob/devel/.github/workflows/reusable-project-builder.yml).  Add a  new reusable workflow that enables FIT testing.  This workflow should have the following parameters:

1. GDS argument: arguments to supply to the GDS launch script and FIT testing script
2. (optional) Run script: path to script used launch binary on target hardware
  1. Default should just be "run" with default IP address/port
  2. Default covers the "we are on target hardware" case (e.g. running on native host)
3. (optional) Clean-up: path to script to clean-up after run
  1. Default should be no-op

With the above arguments to the workflow, the following steps should be run:

1. Download archived build from [project builder](https://github.com/nasa/fprime/blob/devel/.github/workflows/reusable-project-builder.yml). Note: archival may need to be added there
2. Start GDS (in background process)
3. Run "fit" testing script.  Note: may need to add "wait for dataflow" function to FIT framework.
3. Run "run" script (above) to start FSW on target host
4. Wait for FIT script to finish.  This dictates success/failure of the workflow.
5. Run "Clean-up" script (above)
6. Archive results (logs, test runs)

## Rationale

We have a reusable project build script, but not one that does FIT testing.  The above workflow is a generic "run FIT" which has a run/clean-up input used to abstract various hardware platforms.

LeStarch, 2024-03-25:
@thomas-bc would you weigh-in here? Let me know if this general design makes sense as an extension to your designs.

thomas-bc, 2024-06-07:
>3. Run "fit" testing script. Note: may need to add "wait for dataflow" function to FIT framework.
4. Run "run" script (above) to start FSW on target host

Why not switch the order on those 2? I guess with the wait() it wouldn't matter too much, but running FSW before trying to run FIT sounds more natural to me

Otherwise looks good to me!

Note: @rlcheng for ease of development, I would personally develop a simple FIT testing workflow (**not reusable**), and then abstract what can be abstracted in a reusable workflow. But either way works, your call!

LeStarch, 2024-06-07:
@thomas-bc there is a use case where you want to start a test script, then boot software, then test start-up conditions. Otherwise the order is irrelevant.

closed 2025-10-22 by thomas-bc