nasa/fprime #2614: Add Baremetal Reference to CI

LeStarch, 2024-03-25:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

The goal of this epic issue is to run the [fprime-baremetal-reference](https://github.com/fprime-community/fprime-baremetal-reference) with HIL testing via our Github Actions.  

This breaks down into several steps:
- Set up basic FIT tests for the Baremetal Reference: https://github.com/nasa/fprime/issues/2611
- Create FIT testing reusable workflow: https://github.com/nasa/fprime/issues/2612
- Set up baremetal reference as external workflow: https://github.com/nasa/fprime/issues/2613

capsulecorplab, 2024-03-25:
Currently attempting this on the fprime-workspace-image, but encountering build errors https://github.com/fprime-community/fprime-workspace-image/actions/runs/8407060039/job/23021634265

LeStarch, 2025-01-15:
Need to add to CI in nasa/fprime