nasa/fprime #2615: Proper error messaging when missing installation dependencies

AustinOwens, 2024-03-25:
| | |
|:---|:---|
|**_F´ Version_**|  [v3.4.3](https://github.com/nasa/fprime/tree/v3.4.3)  |
|**_Affected Component_**|  `fprime-util generate` |
---
## Problem Description

When installing F', I was originally missing the `default-jre` dependency. This was difficult to determine since the only error message in response to missing `default-jre` was `-- [fpp-tools] fpp-depend appears corrupt` as shown below:

```
(fprime-venv) [root@dfkd83f:~/MyProject]
# fprime-util generate
[INFO] Generating build directory at: /root/MyProject/build-fprime-automatic-native
[INFO] Using toolchain file None for platform default
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [fpp-tools] fpp-depend appears corrupt.
CMake Error at fprime/cmake/required.cmake:26 (message):
   fpp tools not found. Install with:
    'pip install -r "/root/MyProject/fprime/requirements.txt"'
Call Stack (most recent call first):
  fprime/cmake/FPrime.cmake:13 (include)
  CMakeLists.txt:13 (include)
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
# fprime-util version-check
Operating System: Linux
CPU Architecture: riscv64
Platform: Linux-6.2.2-riscv64-with-glibc2.37
Python version: 3.11.8
CMake version: 3.28.3
Pip version: 24.0
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3
```

## How to Reproduce

1. Don't install `default-jre` (I imagine this could happen with other dependencies as well) 
2. Issue the following commands:

```
cd <fprime_project>/
source fprime-venv
fprime-util generate
```

## Expected Behavior

Instead of having "is corrupt" mask dependency errors, print which dependencies are missing or propagate the error upwards to the user.

LeStarch, 2024-03-26:
Some things to note on this issue (for context):
1. On natively supported systems for FPP (`arm` and `x86`) corrupt packages can sometimes be installed. The "corrupt" error should suggest a reinstallation.
2. On non-native systems for FPP Java is required and we need to de-conflate that into a "missing java" error.

thomas-bc, 2024-03-26:
Fixed by #2620

closed 2024-03-26 by thomas-bc