nasa/fprime #2653: Autogenerate Better Release Documentation

LeStarch, 2024-03-28:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Need to autogenerate release documentation.

capsulecorplab, 2024-03-29:
We currently have [auto-generated release notes](https://docs.github.com/en/repositories/releasing-projects-on-github/automatically-generated-release-notes) configured for the fprime-workspace-image https://github.com/fprime-community/fprime-workspace-image/blob/main/.github/release.yml

thomas-bc, 2024-03-29:
Adding some context:

The way we have our documentation versioned is through our docs branch here: https://github.com/nasa/fprime/tree/docs/auto-documentation/docs
And then exposed here https://nasa.github.io/fprime/

Generating these versioned docs is a manual process, through a couple of scripts in the docs/ folder.

The scope of this ticket is to automated that process.

Although we're currently in an effort to revisit the way we serve documentation, so the result of that effort might affect this issue.

LeStarch, 2024-04-01:
@ashleynilo assigned to you because you are working on the whole webpage rework task!

thomas-bc, 2024-12-10:
New website is taking care of this! Woohooo

https://fprime.jpl.nasa.gov

closed 2024-12-10 by thomas-bc