nasa/fprime #2656: fpp-to-xml unable to compile telemetry limits with boolean types

kubiak-jpl, 2024-04-02:
| | |
|:---|:---|
|**_F´ Version_**| v3.4.3 |
|**_Affected Component_**| fpp-to-xml  |
---
## Problem Description

Unable to use telemetry limits with boolean types. The `fpp-to-xml` script is unable to generate an XML file. Here is an example built from the `Ref` deployment

Version 1:
```
telemetry test2: bool id 7 \
low { \
  orange false \
}
```

```
fpp-to-xml
/Users/kubiak/Documents/fprime/Ref/SignalGen/Telemetry.fppi:26.10
  orange false \
         ^
  included at /Users/kubiak/Documents/fprime/Ref/SignalGen/SignalGen.fpp:69.13
error: cannot convert bool to a numeric type
```

Version 2:
```
telemetry test2: bool id 7 \
low { \
  orange 0 \
}
```

```
fpp-to-xml
/Users/kubiak/Documents/fprime/Ref/SignalGen/Telemetry.fppi:26.10
  orange 0 \
         ^
  included at /Users/kubiak/Documents/fprime/Ref/SignalGen/SignalGen.fpp:69.13
error: cannot convert Integer to bool
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Darwin
CPU Architecture: arm64
Platform: macOS-14.4.1-arm64-arm-64bit
Python version: 3.12.2
CMake version: 3.27.4
Pip version: 24.0
[WARNING] Cannot import 'pkg_resources'. Will not check tool versions.
```

Also observed on
```
Operating System: Linux
CPU Architecture: aarch64
Platform: Linux-6.5.0-26-generic-aarch64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a5
```

## How to Reproduce

1. Add the following line to `SignalGen.fpp`
```
telemetry test2: bool id 7 \
low { \
  orange false \
}
```

```
fpp-to-xml
/Users/kubiak/Documents/fprime/Ref/SignalGen/Telemetry.fppi:26.10
  orange false \
         ^
  included at /Users/kubiak/Documents/fprime/Ref/SignalGen/SignalGen.fpp:69.13
error: cannot convert bool to a numeric type
```

2. Rebuild the `Ref` deployment. Observe the following error

## Expected Behavior

Expect to be able to use telemetry limits with Boolean telemetry types

bocchino, 2024-04-29:
This isn't a bug; it's in the language specification. Limits are for numeric types only. In particular, it is not clear what a low or high limit should mean for a Boolean value. Boolean values are not ordered. Consider switching the output to a numeric type if you want to define limits for it.

kubiak-jpl, 2024-05-01:
Maybe this a feature request then. I do think there would be value in telemetry limits for non-numeric types. I don't think of these as numeric limits so much as warnings to the GDS user that a telemetry value requires attention. In that sense it would make sense for boolean and enumeration channels to have channel limits.

bocchino, 2024-05-01:
Sure, we could add the feature of specifying red, orange, and yellow *values* (as opposed to limits) for any telemetry channel type. Can you open a discussion topic on this, to get input from others? Then we can close this issue.

LeStarch, 2025-02-04:
Is this going away with the removal of XML?

kubiak-jpl, 2025-02-04:
I originally wrote the issue because I thought it was an issue with the fpp-to-xml script, but I'm not sure the GDS or fpp supports boolean types. I can open a discussion like Rob recommended if that's the right path forward

LeStarch, 2025-02-04:
I think both the GDS and FPP support booleans. Confirm with Rob, but hopefully we can close.

bocchino, 2025-02-04:
Right, I believe this is a feature request to extend the telemetry colors so they apply to telemetry values that are not numbers. This would require changes to FPP. For example, we would need a way to specify sets of values and match them with colors.

bocchino, 2025-02-04:
Yes, enumeration types are convertible to numeric types, so you can do this:
```
module Fw {

  port Time

  port Tlm

}

enum E { A, B, C }

passive component C {

  time get port timeGetOut

  telemetry port tlmOut

  telemetry T: E low { red E.A } high { red E.B }

}
```
I'm not sure if this behavior has ever been tested in the GDS, though.

bocchino, 2025-02-05:
We have an open issue in FPP to look at colors for enum telemetry output: https://github.com/nasa/fpp/issues/133. Can we close this issue in favor of that one?

kubiak-jpl, 2025-02-05:
That's fine with me

closed 2025-03-19 by bocchino