nasa/fprime #2657: fprime-gds: default file downlink directory can overwrite files

timcanham, 2024-04-03:
| | |
|:---|:---|
|**_F´ Version_**|[v3.4.3](https://github.com/fprime-community/fprime-gds/releases/tag/v3.4.3) |
|**_Affected Component_**| n/a |
---
## Feature Description

Have GDS created a time-tag directory for file downlinks.

Make the default be `logs/<time tag>/fprime-downlink` to match telemetry and events.

Also, make a separate argument for uplink and downlink. It makes sense to automatically separate downlinked files, but uplink may wish to be in a common store.

## Rationale

The default `/tmp/username` (or even overridden with `--file-storage-directory`) directory for file downlinks can cause new instances to overwrite older files, or cause parallel instances to clobber each other.

Joshua-Anderson, 2024-04-03:
Also discussed in https://github.com/nasa/fprime/issues/2658#issuecomment-2035578624

thomas-bc, 2025-04-02:
Closing this issue and tracking the feature request as part of https://github.com/nasa/fprime/issues/2911

closed 2025-04-02 by thomas-bc