nasa/fprime #2658: fprime-gds: file downlink removes leading directories

timcanham, 2024-04-03:
| | |
|:---|:---|
|**_F´ Version_** | [v3.4.3](https://github.com/fprime-community/fprime-gds/releases/tag/v3.4.3) |
|**_Affected Component_**|  n/a |
---
## Problem Description

Add destination directory back in some form. Having raw names can cause overwrites.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
(p3fprime) tcanham@MT-509434:~/source/fprime/Ref$ fprime-util version-check
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a5```

## How to Reproduce

1. Downlink a file with a destination with a directory path
2. Look at file name. Is something like `._Foo.bar`

## Expected Behavior

Normalize the file path and create the directory path to retain directory separation. If it's not desirable to create directories, perhaps use a notation like `dir_path_file` to preserve the separation.

Joshua-Anderson, 2024-04-03:
It's also not uncommon to downlink the same file multiple times, so it would be nice to support that to. (Ex. Run a sequence that generates a logfile)

What I might recommend is:

- Store downlinked files locally in the GDS as `<GDS_DL_TMP_DIR>/<random string>-<filename>`
- In the UI, show their filename as the full filepath
- When downloading, set the filename to be the normal filename

thomas-bc, 2025-04-01:
Confirmed. The current behavior is that it'll replace all slashes `/` by underscores `_`
This was most likely done for security purposes.

https://github.com/nasa/fprime-gds/blob/bfdf5b6a1998a2bb73c515a2877cc904e0822020/src/fprime_gds/common/files/downlinker.py#L203-L212

This seems like a reasonable behavior to me. The description of the command argument is 
```
destFileName: The name of the destination file on the ground
```
which does suggest paths are not allowed.

LeStarch, 2025-04-02:
This is indeed done for security reasons.  Allowing paths is dangerous.  Allowing paths but forcing them to be "relative under the downlink structure" is error prone.  Thus all downlinks are confined to a flat file system.

LeStarch, 2025-04-02:
The current behavior does mirror: dir_path_file functionality requested.  We could prepend a date to the files doenlinked (e.g. file.2025-01-01T10:00:00.000

thomas-bc, 2025-04-02:
Closing this issue and will track the feature request of timestamping directories / files in https://github.com/nasa/fprime/issues/2911

closed 2025-04-02 by thomas-bc