nasa/fprime #2669: F´ GDS Error in Error "TypeError: sequence item 0: expected str instance, PosixPath found"

LeStarch, 2024-04-09:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

When an error arises within the application running code, it triggers another error `TypeError: sequence item 0: expected str instance, PosixPath found` that is reported.

This is caused by not converting `PosixPath` to `str` in our error reporting code.

closed 2024-04-24 by LeStarch

LeStarch, 2024-06-10:
This error originates from error handling code. Fixing it in this issue revealed the true error.

You could install the latest GDS. 

1. Clone F´ GDS: `git clone https://github.com/nasa/fprime-gds`
2. Install it: `cd fprime-gds; pip install .`

This will run the development version where this error is fixed.  It will allow reporting of w/e error is tripping this.

secureshadow, 2024-06-10:
@LeStarch Thank you! That fixes the error. I did not know if I had to install the latest version of F prime to solve the bug, but now it is clear. Thank you! :)

LeStarch, 2024-06-10:
To be clear, you are installing the development version of the GDS tool, not all of F´.

This particular crash is happening in the `Ref` application.  The next thing to do is look in the GDS logs (same directory where you ran GDS there will be a `logs` sub directory with time/date folders for each run).  Look through `Ref.log` and see why it is stopping early.

It depends on where you are running the GDS.  If the GDS is inside the container, then the only port forward you need is 5000 for the browser to connect into the container.

If you are running GDS outside the container, and flight software inside, then it becomes more complicated.  It will depend on, which side is the server in the connection.

We should probably move this discussion to a "Discussion" such that it does not get lost within a closed ticket.

secureshadow, 2024-06-10:
Done! I started a discussion in  #2770. Thanks!