nasa/fprime #2678: Sanitizer Disable Does Not Work

LeStarch, 2024-04-12:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Generating with `-DENABLE_SANITIZER_ADDRESS=OFF` does not work.  Only the `--disable-sanitizer` to disable everything.  The culprit is:

https://github.com/fprime-community/fprime-tools/blob/c15253c98ffc816bcca24db43f24c4c070b7c5ff/src/fprime/fbuild/cli.py#L55-L59

> Notice how the sanitizers are forced-on. 

The fix is in several stages:
1. In CMake default to sanitizers should be == `BUILD_TESTING` (default "ON" when testing).
2. Above code should read "if disable" set the ENABLES "OFF"
3. Take no action on disable sanitizers not set

This way the sanitizers are explicitly turned off, or left to defaults.

thomas-bc, 2024-04-15:
Could it also work if in the 4 lines you linked, each value in the `cmake_args` dictionary would be set to ON only if they are not in the dictionary already (so not passed in by the user)?

LeStarch, 2024-04-15:
That could work too.

closed 2024-11-23 by LeStarch