nasa/fprime #2689: Documentation: Standard Patterns

LeStarch, 2024-04-17:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

- Standard practices:
    - Name with `Request`: asynchronous request
    - Name with `Get`: synchronous request

matt392code, 2024-12-27:
[fpp-port-patterns.md](https://github.com/user-attachments/files/18256362/fpp-port-patterns.md)
# F Prime Port Naming Patterns

## Standard Naming Conventions

### Asynchronous Ports
- Use suffix `Request` for asynchronous operations
- Indicates the operation will be processed in the background
- Caller does not wait for completion

Example patterns:
```fpp
@ Request component to process data
async command processDataRequest(
    ref data: Buffer
) -> ProcessStatus

@ Request state update
async command updateStateRequest(
    state: ComponentState
) -> UpdateStatus

@ Request initialization
async command initRequest() -> InitStatus
```

### Synchronous Ports
- Use prefix `get` for synchronous operations
- Indicates immediate response
- Caller waits for completion

Example patterns:
```fpp
@ Get current component status
sync command getStatus() -> ComponentStatus

@ Get configuration parameter
sync command getParameter(
    id: ParameterId
) -> ParameterValue

@ Get telemetry value
sync command getTelemetry(
    channel: TelemetryId
) -> TelemetryValue
```

## Common Use Cases

### Data Operations
```fpp
@ Async data processing
async command processRequest(
    data: Buffer
) -> ProcessStatus

@ Sync data retrieval
sync command getData(
    id: DataId
) -> Buffer
```

### State Management
```fpp
@ Async state change
async command changeStateRequest(
    newState: ComponentState
) -> StateChangeStatus

@ Sync state query
sync command getState() -> ComponentState
```

### Configuration
```fpp
@ Async configuration update
async command updateConfigRequest(
    config: ConfigBuffer
) -> ConfigStatus

@ Sync configuration retrieval
sync command getConfig() -> ConfigBuffer
```

## Best Practices

1. **Consistency**
   - Always use `Request` suffix for async ports
   - Always use `get` prefix for sync query ports
   - Keep naming consistent across components

2. **Clarity**
   - Names should clearly indicate operation type
   - Use descriptive action verbs
   - Avoid ambiguous terms

3. **Operation Type**
   - Use async ports for:
     - Long-running operations
     - Background processing
     - Operations that might block
   - Use sync ports for:
     - Immediate queries
     - Simple state checks
     - Quick data retrieval

4. **Response Handling**
   - Async ports often return status/acknowledgment
   - Sync ports return requested data directly

## Examples in Context

### Command Component
```fpp
@ Command handling ports
port CommandPorts {
    @ Async command dispatch
    async command dispatchCommandRequest(
        opcode: CommandId,
        args: Buffer
    ) -> CommandStatus

    @ Sync command validation
    sync command getCommandValidation(
        opcode: CommandId
    ) -> ValidationStatus
}
```

### Telemetry Component
```fpp
@ Telemetry handling ports
port TelemetryPorts {
    @ Async telemetry update
    async command updateTelemetryRequest(
        channel: ChannelId,
        value: TelemetryValue
    ) -> UpdateStatus

    @ Sync telemetry query
    sync command getTelemetryValue(
        channel: ChannelId
    ) -> TelemetryValue
}
```

### Parameter Component
```fpp
@ Parameter management ports
port ParameterPorts {
    @ Async parameter update
    async command setParameterRequest(
        id: ParameterId,
        value: ParameterValue
    ) -> SetStatus

    @ Sync parameter query
    sync command getParameterValue(
        id: ParameterId
    ) -> ParameterValue
}
```

## Common Patterns and Usage

### Request-Response Pattern
```fpp
@ Async request with later response
port RequestResponsePorts {
    @ Initial request
    async command startOperationRequest(
        params: OperationParams
    ) -> RequestStatus

    @ Get operation results
    sync command getOperationResults(
        requestId: RequestId
    ) -> OperationResults
}
```

### Update-Query Pattern
```fpp
@ Update and query pattern
port UpdateQueryPorts {
    @ Async update
    async command updateDataRequest(
        data: UpdateBuffer
    ) -> UpdateStatus

    @ Sync query
    sync command getData() -> DataBuffer
}
```

## Verification Checklist

When reviewing port definitions, verify:

1. **Naming Convention**
   - [ ] Async ports end with `Request`
   - [ ] Sync query ports start with `get`
   - [ ] Names are clear and descriptive

2. **Port Type**
   - [ ] Async used for long operations
   - [ ] Sync used for immediate queries
   - [ ] Type matches operation nature

3. **Parameters and Returns**
   - [ ] Parameters clearly named
   - [ ] Return types appropriate
   - [ ] Status types for async operations

4. **Documentation**
   - [ ] Clear port description
   - [ ] Parameters documented
   - [ ] Return values explained

LeStarch, 2025-01-15:
This will be done as part of #3115.  Closing as duplicate!

closed 2025-01-15 by LeStarch