nasa/fprime #2694: Tutorial extension - UTs and ITs

rlcheng, 2024-04-17:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

* Unit Tests
* Integration Tests

Possibly as a part 2 to the Helloworld tutorial.

## Rationale

Give new users a guide on how UTs and ITs are done in F Prime.

Joshua-Anderson, 2024-04-18:
We have UT and IT tests guides included as part of the LED blinker tutorial. Is there value to adding another UT/IT guide to helloworld? The goal of the helloworld tutorial in my mind is to be the minimal viable fprime example. The led blinker tutorial is the next step and starts to add some more complicated functionality

ljreder, 2024-04-18:
@Joshua-Anderson I personnelly like the UT example in the Math Tutorial but have not been through the LED tutorial yet.   I think the helloworld tutorial is just fine an excellent starting point.  

My problem has been in the Math Tutorial I have not been able to get the unit test to work even though all the user code is correct and matching the reference repo.  I am not sure if it is just a version problem but issue #2514 is reopened since after trying it again from scratch the unit test failed.

@rlcheng any chance you can take a look and try out the Math Tutorial on a fresh project and virtual environment and see if it works for you?

rlcheng, 2024-04-18:
@Joshua-Anderson when @LeStarch and I were talking about the thinking was doing a part 2 to HelloWorld so a new user can choose to run through it or not.

I didn't realize that the LED blinker tutorial has UTs and ITs already but I think it's ok to introduce the concept more than once, especially one where I don't need additional setup/hardware (which is why I haven't done the LED blinker, haven't gotten around to dusting off my Pi Zero and LEDs).

Perhaps after someone finishes all the tutorials, they come away with: test early and test often.

@ljreder I'll take a look at the Math Tutorial soon.

Joshua-Anderson, 2024-04-18:
@rlcheng the LED blinker tutorial explicitally does not require hardware - maybe that's something we should better document.

Fprime is always evolving, and one of the big challenges I've experienced is keeping the documentation up to date. Out of date and incorrect tutorials can lead to lots of confusion and folks giving up on fprime. In my view (and I invite others to disagree!) I'd rather have a smaller number of really good guides and tutorials to key concepts like unit and integration tests, that way the fprime infrastructure team has a fair chance of running through and catching issues as things break over releases.

I'm not sure folks would learn anything different from writing UT/IT for hello world vs the blinking tutorials... so wonder if it's better for us to pick one repository to demo and walk through UT/ITs.

ljreder, 2024-04-18:
Absolutely agree!  THANKS @Joshua-Anderson.

rlcheng, 2024-04-18:
@Joshua-Anderson thank you. I like your idea of smaller number of really good guides and tutorials.

Maybe we need to think of this as: if we were to do a week long class, what key concepts / tutorials do we need each day and structure our guides / tutorials that way.

LeStarch, 2025-02-03:
I am going to close this as the discussion points out that these are demonstrated in the larger tutorial set.  Reopen if I missed something.

closed 2025-02-03 by LeStarch