nasa/fprime #2706: Failures of TCP/UDP unit tests

JohanBertrand, 2024-04-28:
| | |
|:---|:---|
|**_F´ Version_**| 6bb780d |
|**_Affected Component_**|   |
---
## Problem Description

Similar to https://github.com/nasa/fprime/issues/1290, https://github.com/nasa/fprime/issues/881

Maybe also related to https://github.com/nasa/fprime/issues/2244

I am not sure if it is having an impact outside of the UT.

The tests of `Drv/TcpServer` and  `Drv/TcpClient` are often failing, about 25% of the time (excluding silent failures).

It comes from several different errors:
- For TcpServer and TcpClient, most of the time, an error comes from the Address already being in use (Port already in use ?), based on `errno`. Also happens on the Github CI: https://github.com/nasa/fprime/actions/runs/8839296509/job/24272334136
```
[ RUN      ] Nominal.BasicMessaging
TCP server startup error: Address already in use
Port: 48514
/home/fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:49: Failure
Expected equality of these values:
  serverStat
    Which is: -9
  SOCK_SUCCESS
    Which is: 0
[  FAILED  ] Nominal.BasicMessaging (0 ms)
```
```
[ RUN      ] Reconnect.ReceiveThreadReconnect
TCP server startup error:Address already in use
Port: 33260
/home/fprime/Drv/TcpClient/test/ut/TcpClientTester.cpp:44: Failure
Expected equality of these values:
  serverStat
    Which is: -9
  SOCK_SUCCESS
    Which is: 0
[  FAILED  ] Reconnect.ReceiveThreadReconnect (0 ms)
```
- Sometimes TcpServer UT fail and can not start:
```

[ RUN      ] Nominal.BasicReceiveThread
[WARNING] Failed to open port with status -9 and errno 98
[WARNING] Failed to open port with status -9 and errno 98
/home/fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:41: Failure
Value of: Drv::Test::wait_on_started(this->component.getSocketHandler(), true, SOCKET_RETRY_INTERVAL_MS/10 + 1)
  Actual: false
Expected: true
/home/fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:51: Failure
Value of: component.getSocketHandler().isStarted()
  Actual: false
Expected: true
[WARNING] Failed to open port with status -9 and errno 98
/home/fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:65: Failure
Value of: Drv::Test::wait_on_change(this->component.getSocketHandler(), true, SOCKET_RETRY_INTERVAL_MS/10 + 1)
  Actual: false
Expected: true
/home/fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:67: Failure
Value of: this->component.getSocketHandler().isOpened()
  Actual: false
Expected: true
/home/fprime/Drv/TcpServer/test/ut/TcpServerTester.cpp:70: Failure
Expected equality of these values:
  status2
    Which is: -4
  Drv::SOCK_SUCCESS
    Which is: 0
[... 192 lines cut ...]
    executable = call.split('/')[-1]
    print(executable)
    for ii in range(0, ITERATIONS):
        print(f"Test number {ii:04}")
        try:
            output = subprocess.check_output([call], cwd=FPRIME_PATH, stderr=subprocess.PIPE)
            string_output = output.decode()
            if "[WARNING] Failed" in string_output:
                print("Silent failure")
                f = open(LOG_PATH + f"silent_{executable}_{ii:04}.log", "x")
                f.write(string_output)
                f.close()
        except subprocess.CalledProcessError as grepexc:
            print("Error code:", grepexc.returncode)
            f = open(LOG_PATH + f"{executable}_{ii:04}.log", "x")
            f.write(grepexc.output.decode())
            f.close()
```
5. Tune the values `FPRIME_PATH` to the current fprime folder and `LOG_PATH` to an empty folder to save log of unit tests. `ITERATIONS` can also be changed to increase the amount of run of the unit tests
6. Execute the python script
7. Check the log in the `LOG_PATH` folder

## Expected Behavior

No error should be generated. The unit tests should also fail on warning

JohanBertrand, 2024-05-23:
https://github.com/nasa/fprime/pull/2739 fixed the non-silent errors. The silent warnings are still here

thomas-bc, 2024-10-16:
Fixed by https://github.com/nasa/fprime/pull/2950

closed 2024-10-16 by thomas-bc