nasa/fprime #2714: Consider having `Fw::Buffer::getSerializeRepr` return an object instead of a reference

bocchino, 2024-05-01:
`Fw::Buffer::getSerializeRepr` currently returns a reference into a serialize buffer stored as a member of `Fw::Buffer`. There are some issues with this:

1. When you call `getSerializeRepr`, you're not sure what state the serialize buffer is in. It depends on what has previously been done with the buffer.
2. The serialize representation state is reset when the buffer object is serialized and deserialized but otherwise persists. This could be an unexpected behavior.
3. Performing deserialize operations on the buffer updates the serialize buffer and therefore updates the `Fw::Buffer`. It would be better for the`Fw::Buffer` to be read-only in this case.

Instead we could have `getSerializeRepr` return an object, say `Fw::ExternalSerializeBufferWithMemberCopy`. That way the user would own the serialize representation, and it could be freshly initialized each time it is requested. Access to the buffer would be read-only. An argument to `getSerializeRepr` could specify whether the representation is for serialization (in which case the internal serial pointer should be set to zero) or deserialization (in which case the internal serial pointer should be set to the end of the data).

LeStarch, 2024-05-01:
I like it!  Thanks for the write-up @bocchino!

LeStarch, 2025-03-26:
We should do:
1. Separate serialize/deserialize methods. (e.g. `getSerializer` and `getDeserializer`) using Robs suggestions above.
2. Deprecate (`Fw/Deprecate.hpp`) the `getSerializeRepr` method, but don't remove the functionality (yet)

thomas-bc, 2025-04-15:
Implemented by #3431  🎉

closed 2025-04-15 by thomas-bc