nasa/fprime #2718: HelloWorld Tutorial - Deployment Build Error

reporterfredbriggs, 2024-05-06:
| | |
|:---|:---|
|**V3.4.3**| |
|**Hello World**|   |
---
## Problem Description

After following the intro documentation pages up to this page https://fprime-community.github.io/fprime-tutorial-hello-world/docs/Deployments.html and then trying to execute the command: 

```
# In: MyProject/HelloWorldDeployment
fprime-util build -j4
```

to try to set up and build the deployment i'm getting these errors:
```

make[3]: *** No rule to make target `HelloWorldDeployment/Top/HelloWorldDeploymentTopologyAppAi.xml', needed by `Components/HelloWorld/HelloWorldDeployment/Top/HelloWorldDeploymentPacketsAc.cpp'.  Stop.
make[2]: *** [Components/HelloWorld/HelloWorldDeployment/Top/CMakeFiles/Components_HelloWorld_HelloWorldDeployment_Top.dir/all] Error 2
make[1]: *** [Components/HelloWorld/HelloWorldDeployment/CMakeFiles/Components_HelloWorld_HelloWorldDeployment.dir/rule] Error 2
make: *** [Components_HelloWorld_HelloWorldDeployment] Error 2
[ 97%] Built target Svc_FileUplink
[ 97%] Built target Svc_Framer
[ERROR] CMake erred with return code 2
```

## Context / Environment

```
Operating System: Darwin
CPU Architecture: x86_64
Platform: macOS-12.5.1-x86_64-i386-64bit
Python version: 3.12.3
CMake version: 3.29.2
Pip version: 24.0
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3
```

## How to Reproduce

1. Start with the HelloWorld Tutorial on https://fprime-community.github.io/fprime-tutorial-hello-world/
2. Create the project following: [Creating an F´ Project](https://fprime-community.github.io/fprime-tutorial-hello-world/docs/NewProject.html)
3. Then [Creating an F´ Hello World Component](https://fprime-community.github.io/fprime-tutorial-hello-world/docs/HelloWorld.html)
4. In the "Creating an F' Hello World Component, try the # In: MyProject/HelloWorldDeployment
fprime-util build -j4 command

## Expected Behavior

It should built without error.

thomas-bc, 2024-05-06:
@reporterfredbriggs 
in the error message: `Components/HelloWorld/HelloWorldDeployment/`

You seem to have created the `HelloWorldDeployment` within your `Components/HelloWorld`, which is why you're getting an error.
You should start fresh and make sure to `cd` back at the root of the project before running `fprime-util new --deployment`
It is mentioned in the tutorial as a bash comment (see below)... it should probably be made more clear...
```
# In: MyProject
fprime-util new --deployment
```

If you would be willing to, please make a contribution to the tutorial repo to reword things or add a note to make sure this is seen by users, this would be a really good contribution, because we've seen this issue reported in the past.

thomas-bc, 2024-05-06:
Closing this issue, let us know if you run into other issues.

closed 2024-05-06 by thomas-bc

reporterfredbriggs, 2024-05-07:
Hi Thomas, the second approach worked, but then gave me another error at 98% build. Going to try your first resolution approach to see how that changes things. Thanks for the quick response time! I'll look into adding to the repo once I get this sorted.