nasa/fprime #2724: Clean Out Non-Standard OSAL Layer Items

LeStarch, 2024-05-10:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Remove `Os::Watchdog` as it is VxWorks specific without clear pattern for other OSes.

menaman123, 2024-05-26:
Hi, I was wondering if Im able to do this task. This would be my first contribution to this project or any open source project.

aramk-hub, 2024-09-06:
Hey, I was wondering if this was still in progress / being worked on? I'm looking for my first contribution to F' right now. If this isn't being worked on anymore, I'm happy to take over. Otherwise, I see some first issues open, but what do you suggest to work on out of that list?

aramk-hub, 2024-09-06:
Hey, I was wondering if this was still being worked on or in progress. If not, I'm happy to take over. Otherwise, I do see some first-issue labels open, but what specifically would be a good issue to gain exposure to the core fundamentals? I've loved space for a long time and found this, so I'm excited and want to make sure I have a solid foundation to contribute often.

aramk-hub, 2024-09-06:
Hey @LeStarch, sorry for tagging you, I'm not sure if you get notifications for every response, but it seems like you were the lead in this issue. Thanks for any help!

LeStarch, 2024-09-06:
This is not underway yet.  It will be done by our team in 2 weeks or so.....but if you want to do it first, go for it!

aramk-hub, 2024-09-08:
Sure I'll give it a shot! Looking at `Os::Watchdog` right now and seeing the Timer classes, I assume this is what we want to remove, is that correct? Also, does this issue need all the references of `Os::Watchdog` deleted as well (i.e. in the `os-docs.md` or `CMakeLists.txt, etc.)?

LeStarch, 2024-09-23:
Preferably we'd remove documentation references, yes.

closed 2024-10-10 by LeStarch