nasa/fprime #2727: Refactor Os::SystemResources into CMake Selection

LeStarch, 2024-05-10:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

- [ ] Refactor Os::Log into the Interface, Os::Log, and Implementation classes (following pattern of Os::File)
- [ ] Implement Posix implementation
- [ ] Implement Stub implementation
- [ ] Implement Test stub implementation
- [ ] Build common rules-based tests
- [ ] Build specific posix tests
- [ ] Build stub interface tests
- [ ] Setup default posix implementation for Posix platforms
- [ ] Document

thomas-bc, 2024-06-13:
Note: 
-> use singleton pattern in the "face" class (`class SystemResources final : public SystemResourcesInterface {` to which the static methods delegate to the singleton

LeStarch, 2024-10-01:
@kevin-f-ortega reassigning this to me.

closed 2024-10-09 by thomas-bc