nasa/fprime #2730: FPrime GDS does not gracefully handle files downlinks of the same filename

rrieber, 2024-05-13:
Say an FPrime spacecraft downlinks `filename.bin`, then downlinks the same file a second time for redundancy. The current behavior of FPrime GDS would be to overwrite the first `filename.bin` with the second. If there were undetected corruption or other issues, those would be permanently imprinted on the final version of the file.

Let's explore another scenario. Say an FPrime spacecraft downlinks `image.bin`. Then deletes it, makes a new `image.bin` and downlinks that. FPrime GDS would do the same thing, it would overwrite the first version of `image.bin` with the second version if `image.bin`. 

Note that both of these operational scenarios are used all the time. I'm not talking once per mission, I'm talking multiple times per day.

I think FPrime GDS should be modified to ensure received files all have unique names. I've seen them appended with the Earth Received Time (ERT) with great success. It's easy to remove with a simple RegEx and ensures uniqueness.

See also #2458 for a mildly-related issue.

Joshua-Anderson, 2024-05-14:
The reason the GDS is currently designed this way is to properly handle partial file downlinks (i.e. `Filedownlink.SendPartial`). You can do multiple partial downlinks, appending/updating the existing file.

I wouldn't be overly worried about file corruption, since all file packets are checksummed, but if you're overwriting an existing file onboard the spacecraft, combining the files on the ground can definitely lead to confusion.

Since the fprime-gds is primarily for ground testing, I think designing around user friendlyness and appending ERT to the filename of downlinked files makes sense to me. Occasional users of SendPartial can manually append files together.

@thomas-bc What do you think about:

- On disk, storing the downlinked file as `ERT-filename`
- In the GDS file downlink GUI, showing just the filename, and users can click on the exact version of the file in the downlink list they want to download.

thomas-bc, 2024-05-14:
That approach makes sense to me!

LeStarch, 2025-02-03:
We need to answer if F Prime should distinguish between multiple downlinks of the same file, or if it should pass to the project.

Also, is this superseded by CFDP?

thomas-bc, 2026-01-16:
CCB: let's go with the ERT/timestamp approach. When downlinking a file, add a timestamp to the file saved on disk. That solves the issue with the downside that partial downlinks impose on the user to manually merge the files. Less user friendly in some cases, but more robust. Need to document clearly, and potentially provide a merge script for the ground operators.

thomas-bc, 2026-09-09:
CFDP has been delivered in v4.3 - closing this.

closed 2026-09-09 by thomas-bc

keck-in-space, 2026-09-10:
How does CFDP approach this issue? Are file names unique on disk?

thomas-bc, 2026-09-10:
CFDP probably doesn't say anything about this so it is up to the GDS implementation. F prime GDS does not plan on supporting CFDP at this time - we should look into what YAMCS does about it.

rrieber, 2026-09-10:
@keck-in-space In the AMMOS version of CFDP GDS, the telemetered files are stamped with the CFDP transaction ID, which is a flight system implementation to ensure is unique. Historically, projects have done things like <APID>_<SCLK>-<subsclk>.<extension>. Hope this helps.