nasa/fprime #2756: Attempting to Downlink Zero-Length File Results in Poor Warning

Joshua-Anderson, 2024-06-03:
| | |
|:---|:---|
|**_F´ Version_**| v3.4.3 |
|**_Affected Component_**| Filedownlinke   |
---
## Problem Description

Attempting to downlink a zero length file using FileDownlink results in the following warning

fileDownlink.DownlinkPartialFail | WARNING_HI | Error occurred during partial downlink of file badfile to file badfile. Offset 0 greater than or equal to source filesize 0.
-- | -- | --

## How to Reproduce

Send `fileDownlink.SendFile, "badfile", "badfile"`, where badfile is zero length

## Expected Behavior

Send a specific warning EVR instead of telling a user they had an incorrect file partial command.

I would propose

fileDownlink.DownlinkEmptyFile | WARNING_HI | Cannot downlink empty file %s. 
-- | -- | --

LeStarch, 2024-06-04:
Is it illegal to downlink an empty file?  Perhaps the issue is that we are erring on an empty file instead of properly processing it.

Joshua-Anderson, 2024-06-04:
Properly supporting downlinking zero length files is also a valid approach. That being said, I don't think there's any information conveyed by downlinking a empty file since we're not sending any file metadata. Sending a warning EVR and skipping transmitting the actual file downlink packets may help flag operators that something is anomalous, since I haven't been able to think of a case where folks would intentionally downlink an empty file.

LeStarch, 2024-06-04:
I thought the start packet had the length (which would be 0).  This would be for the case of needing to downlink a file (e.g. a log file) that happens to be empty.

Joshua-Anderson, 2024-06-04:
From the ground side, yes, it would look like this:

```
file downlink start packet <size> <src name> <dest name>
file downlink end packet <checksum
```

So the only information you would receive from transferring a zero length file is that the file size is zero.

Sending a file is zero length warning EVR would have the benefit of both providing the information that the file is zero length while also providing a warning to operators that something anomalous occurred.

If there's cases where somebody would intentionally downlink a zero length file - I'd lean towards allowing downlink of zero length files. But I haven't been able to think of any intentional cases, so I'm slightly leaning towards sending a warning event instead of sending the empty file.

LeStarch, 2025-02-03:
We need to determine if it is valid to downlink and empty file.  If so, it should work.  If not, it should error.