nasa/fprime #2760: "bad interpreter: Permission denied" error when building on a file path that contains spaces

SiboVG, 2024-06-06:
| | |
|:---|:---|
|**_F´ Version_**| fprime-tools==3.4.4, fprime-gds==3.4.3, fprime-fpp-*==2.1.0a3|
|**_Affected Component_**| fprime-util build |
---
## Problem Description

I'm following the Hello World example and am [at this section](https://fprime-community.github.io/fprime-tutorial-hello-world/docs/Deployments.html#:~:text=fprime%2Dutil%20build%20%2Dj4). However, when I run the `fprime-util build -j4
` command, I get this error:

```bash
$ fprime-util build -j4     

[  0%] Generating <my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime/Autocoders/Python/src/fprime_ac/generators/templates/arrays/array_cpp.py, <my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime/Autocoders/Python/src/fprime_ac/generators/templates/arrays/array_hpp.py
[  0%] Generating <my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime/Autocoders/Python/src/fprime_ac/generators/templates/channels/ChannelBody.py, <my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime/Autocoders/Python/src/fprime_ac/generators/templates/channels/ChannelHeader.py
[  1%] Built target config
/bin/sh: <my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime-venv/bin/cheetah-compile: <my_home_dir>/Flight: bad interpreter: Permission denied
/bin/sh: <my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime-venv/bin/cheetah-compile: <my_home_dir>/Flight: bad interpreter: Permission denied
make[2]: *** [<my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime/Autocoders/Python/src/fprime_ac/generators/templates/channels/ChannelBody.py] Error 126
make[2]: *** Waiting for unfinished jobs....
[  1%] Generating <my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime/Autocoders/Python/src/fprime_ac/generators/templates/commands/CommandBody.py, <my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime/Autocoders/Python/src/fprime_ac/generators/templates/commands/CommandHeader.py
make[2]: *** [<my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime/Autocoders/Python/src/fprime_ac/generators/templates/arrays/array_cpp.py] Error 126
/bin/sh: <my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime-venv/bin/cheetah-compile: <my_home_dir>/Flight: bad interpreter: Permission denied
make[2]: *** [<my_home_dir>/Flight Software/FPrime_Sandbox/HelloWorld/fprime/Autocoders/Python/src/fprime_ac/generators/templates/commands/CommandBody.py] Error 126
make[1]: *** [F-Prime/Autocoders/Python/src/fprime_ac/generators/templates/CMakeFiles/codegen.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  1%] Built target Fw_Cfg
[  1%] Built target version
make: *** [all] Error 2
[ERROR] CMake erred with return code 2
```

As you can see, I have one directory in my build path, "Flight Software", which has a space in its name. This causes the error. I know, software engineers will look at me funny because I use directories with space, but in any case, the build process should resistant to this IMO. 

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Darwin
CPU Architecture: arm64
Platform: macOS-14.5-arm64-arm-64bit
Python version: 3.11.9
CMake version: 3.29.2
Pip version: 24.0
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3
```

## How to Reproduce

1. Create a new project in a path where its (super-)parent has a space in its name
2. Add deployment etc.
3. Try to build the deployment
4. Error...

## Expected Behavior

F' should be robust against file paths that contain a space.

thomas-bc, 2024-06-06:
This is not something that can be handled by CMake nicely. The build system is invoking a python script through CMake, and CMake doesn't escape it.

If not CMake, it will be another tool down the road. 

We are not planning to support this unfortunately. You will need a path that has no space in it. `fprime-bootstrap` should warn against this when creating the project.

SiboVG, 2024-06-06:
> This is not something that can be handled by CMake nicely. The build system is invoking a python script through CMake, and CMake doesn't escape it.
>
> If not CMake, it will be another tool down the [...]

Ah, alright, was not aware of this. Thanks for the reply! I already converted my project to have no spaces in the file path and everything is working now.

closed 2024-06-07 by thomas-bc