nasa/fprime #2763: Break Apart Deframer

LeStarch, 2024-06-07:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

The Deframer is performing several peices:

1. It reassembles frames
2. It calls custom deframing via a subclass
3. It routes data to F´ components.

On modern projects, we have seen the following needs that the deframer cannot support. 

1. Custom deframing needs to use custom port calls (e.g. call out to a helper component, database component, etc)
2. Routing to F´ components often needs to be customized
3. Reassembly is optional (e.g. a packet radio is already produces complete frames)

Thus we will break the deframer into 3 F´ items:

1. An optional accumulator that takes in byte streams and outputs complete frames in Fw::Buffer:
2. A deframer interface that allows custom deframing to a standard component "drop in":
3. A router implementation for basic F´ constructs:

matt392code, 2025-01-11:
Proposed Deframer Interface:
[deframer-interface.txt](https://github.com/user-attachments/files/18383753/deframer-interface.txt)
Proposed basic F Prime router implementation:
[uplink-router(1).txt](https://github.com/user-attachments/files/18383759/uplink-router.1.txt)
Proposed implementation file for the UplinkRouter:
[uplink-router-cpp.txt](https://github.com/user-attachments/files/18383764/uplink-router-cpp.txt)
Let me explain the key aspects of this implementation:
1. DeframerInterface (deframer-interface.hpp):
- Pure virtual interface for custom deframing implementations
- Defines key methods that must be implemented:
  - deframe(): Process incoming frame data
  - getPacketType(): Extract packet type from deframed data
  - validateFrame(): Validate frame before processing
- Status codes for deframing operations
- Clean separation of concerns focusing only on deframing

2. UplinkRouter (uplink-router.hpp and uplink-router.cpp):
- Handles routing of deframed packets to F Prime components
- Supports command and file packet types
- Maintains statistics on routed packets
- Follows F Prime component patterns
- Key features:
  - Command packet routing via ComBuffer
  - File packet routing with proper buffer management
  - Error handling and logging
  - Port connection validation

The design achieves several goals:
1. Separation of concerns between deframing and routing
2. Flexibility for custom deframing implementations
3. Standard interface for F Prime integration
4. Clean error handling and logging
5. Statistics tracking for monitoring

thomas-bc, 2025-03-12:
Closed by #3250  !!! 🎉

closed 2025-03-12 by thomas-bc